nasa/nos3 #2: About VM ID and Passworld

hrkimkau, 2018-09-13:
Hello, All.

I can find pass word but I can't find ID. 

Please let me know about ID.

jlucas9, 2018-09-13:
Hello hrkimkau,

The user ID is `nos3` and the password is `nos3123!`.  As you go through the readme, note that the provisioning process can take quite awhile depending on the speed of your internet connection and your computer.  During provisioning, which includes both vagrant commands, a VM will appear with a black login screen.  Vagrant is actually logged in and setting up the GUI for you behind the scenes.  Once you've dropped into a terminal and performed the `vagrant reload` command the VM will restart and will (pending everything worked properly) present an Ubuntu desktop GUI you can login to.  

Please let me know if you have any more issues and have a great day!

John

jlucas9, 2018-10-17:
Hello ledessu,

My first guess is that the repository was not able to successfully copy into the virtual machine (VM).  Failing at that would also mean the your VM is not going to work for you.  Could you capture the log file during provisioning?  It's easiest to use the `vagrant up > nos3_install.log` at the command line instead of the `vagrant up` command.  You won't be able to see the progress in the terminal, but it will be saved in the log.  

In order to delete your current VM and attempt to get this log from scratch, you'll need to navigate to `nos3/support` and run `vagrant destroy` and confirm the command with `y`.  

Thanks and good luck!

ledessu, 2018-10-18:
Hello jlucas9,
Your response was important to me. Now I re run the vagrant up as you suggested and now it seems working. I am able to build and run nos3 from the desktop. But I am not able to get cosmos. If you have some other materials to support a beginner like me, that would be appreciated.

jlucas9, 2018-10-18:
Hi ledessu,
I apologize for this.  It appears that the latest version of COSMOS requires a new version of the ruby `bundler`.  We have a fix to the provisioning scripts and hope to push it to Git as soon as it is tested.  The fix is to open a terminal in the VM and perform the following:
1.  cd ~/Desktop/cosmos
2.  bundle update
3.  ruby Launcher &
The COSMOS launcher window should appear verifying the fix.  We hope to also publish some additional documentation and a FAQ section in the readme of the next release.  Until then, feel free to create additional issues via GitHub or email `support@nos3.org`.  Have a great day!

closed 2018-10-18 by jlucas9

ledessu, 2018-10-19:
That is great. I will stay tuned.