nasa/nos3 #5: Nos3 Windows Launcher Issue

SamuelChristensen, 2019-02-05:
I am getting the same error as cscase, where a console box comes up and asks for the vagrant login and password. I have restarted the launcher several times after deleting the VM and encounter the same problem every time.

jlucas9, 2019-02-06:
Hi Samuel,
Can you do the following in the terminal / command prompt and post the `output.txt` file?
```
cd nos3/support
vagrant destroy
y # Confirm vagrant destroy
vagrant up > output.txt
# The VM should open to a black login screen quickly,
# but wait until the terminal returns to continue 
vagrant reload
# The VM should now have a nice desktop environment available!
```
Also what is your host OS, Virtual Box version, and Vagrant version?
Thanks!

SamuelChristensen, 2019-02-08:
When running that the terminal does not actually return to let my type vagrant reload into it The vagrant login just shows up again. But I have the output.txt.

I am using Windows 10 with Virtual Box 6.0.2 and Vagrant 2.2.3

[output.txt](https://github.com/nasa/nos3/files/2846202/output.txt)

jlucas9, 2019-02-12:
Hi Samuel,
I just updated my Windows 10 laptop to Virtual Box 6.0.4 and Vagrant 2.2.3 and was able to provision the latest master branch of NOS3 (note this updated since the issue was opened) using the most recent base box.  You can update the base box by doing the following:
```
cd nos3/support
vagrant box update
```
[nos3_20190212.txt](https://github.com/nasa/nos3/files/2856034/nos3_20190212.txt)
Above is my build log I generated following the instructions in my first reply.  I was notified of issues in early versions of Virtual Box 6 which is why I waited until now to update.  See if you can update that and post your log file again.  It took about forty five minutes for my box to provision, but that is very dependent on internet and host PC speeds.
Thanks and sorry this is taking so long to diagnose.  Have a great day!

SamuelChristensen, 2019-02-15:
I updated all of my programs (Virtual Box 6.0.4, Vagrant 2.2.3, and the newest NOS3 master). Executed the vagrant box update. I think the latest update was already installed since it took at most 5 seconds to finish that command. (Here is the command prompt after that)

[image]

Then I reran the code above with vagrant destroy and vagrant up > output.txt
Here is the output file, 
[output.txt](https://github.com/nasa/nos3/files/2870101/output.txt)

It looks the same as the one I posted earlier. And the same problem happened where the login screen appeared (shown below)

[image]

Thanks for the help

jlucas9, 2019-02-18:
Hi again Samual,
The output file doesn't show an error so this is very difficult to diagnose.  My guess is that the provisioning is still going on as this takes forty five minutes on a high end laptop with a gigabit internet connection before you can perform the `vagrant reload`.  In order to make this process easier I've created an OVA that you can download and simply import into Virtual Box - double clicking it should start the process as the file type is most likely already recognized as related to Virtual Box.  The files can be found here in [Release 1.03.01](https://github.com/nasa/nos3/releases/tag/v1_03_01).
Let me know how that goes for you and have a great day!

SamuelChristensen, 2019-03-15:
Sorry for waiting so long to post, but the OVA worked. Thank you so much for helping me :)

closed 2019-03-18 by jlucas9