nasa/nos3 #8: no nos3 in cosmos target

refaat100, 2019-02-17:
Hi,

After installing everything when run the script of nos3-run in cosmos there is no target with nos3 and in the command termonal also.

and in buses selection no simulation for gps and imu ,...

thanks in advance

jlucas9, 2019-02-18:
Hi refaat100,
Are you running the `nos3-run.sh` script in the provisioned Virtual Machine?  
You can find basic installation instructions in the [readme.md](https://github.com/nasa/nos3/blob/1.03.01/README.md) file while details are contained [here](https://drive.google.com/open?id=1sg86WNX_mCH59ZAvQQGN85prnGaCM0AT).  If you are still having issues, let me know and I'll see what I can do to help out.  
Thanks and have a great day!

scsercan, 2019-02-20:
hi, i have same issue. there is no nos3 target on cosmos. I installed package again and again following your readme.md step by step but nothing change. Also there is 5 simulation terminals not 7 when i ran nos3. What can be the problem?

jlucas9, 2019-02-20:
Hi scsercan,
Digging into the `nos3_run.sh` [script](https://github.com/nasa/nos3/blob/master/support/VirtualMachine/scripts/nos3-run.sh#L30) is appears that only five tabs are supposed to be open.  It appears that the documentation was developed using the Simulation To Flight - 1 (STF-1) version of NOS3 which includes mission specific simulators that have not been released.  While in the `NOS Engine Standalone Server` terminal you can press 2 for Data Nodes to show that both the GPS and FSW communicate via `usart_1`.  The camera simulator uses both I2C and SPI interfaces.  
COSMOS also now only has a `MISSION_INT` interface that is utilized instead of `NOS3` since that will be customized by the user.
Please let me know if you have any follow up questions and have a great day!

scsercan, 2019-02-20:
thank you for your answer jlucas9. I've got some errors but i don't know if these are important. I will open issue for them.
Have a great day!

closed 2019-02-21 by jlucas9