nasa/nos3 #13: Login incorrect

lipofefeyt, 2019-03-28:
Hi,

I have tried different setups:
- Windows host -> VMWare running Ubuntu 16.04 x64 Guest
- Windows host -> VirtualBox running Ubuntu 16.04 x64 Guest

I have followed the installation steps until I get the login prompt. I keep getting this "login incorrect" message nevertheless, when trying to login with:
user: nos3
pwd: nos3123!

I have also tried running the setup directly in windows, running the "windows_nos3_installer.bat" script.

Could someone please point out if I'm doing something wrong?

Thanks a lot!

jlucas9, 2019-03-28:
Hi @lipofefeyt,  

The provisioning process using vagrant is to be run on your host directly via a terminal or command prompt.  Vagrant will download a base box for use and run some script that will download all the necessary packages and install everything for you.  This takes quite awhile (minimum thirty minutes max two hours as a rough estimate) depending on your host machine and internet connection.  Once everything has completed and you've done the `vagrant up`, waited what feels like forever, and then `vagrant reload` you should be presented with a Virtual Machine that has a GUI and allows you to login.
  
Hopefully this helps!  You can always load the OVA provided by a [release](https://github.com/nasa/nos3/releases) if you don't want to install vagrant and go through the provisioning process.

lipofefeyt, 2019-03-29:
Hi @jlucas9,

Thanks for your very quick response. I will try it on my side and I'll come back to you with some feedback.

Cheers!

jlucas9, 2019-07-15:
Closing - let me know if you have any other issues!

closed 2019-07-15 by jlucas9