nasa/nos3 #20: Same problem installing gem

seanjone, 2019-06-04:
default: ruby 2.3.1p112 (2016-04-26 revision 54768) [x86_64-linux]
    default: chown: cannot access '/var/lib/gems'
    default: : No such file or directory
    default: Gem installation...
    default: mkdir: cannot create directory ‘/home/root/Desktop’: No such file or directory
    default: bash: line 7: /home/root/.gemrc: No such file or directory
    default: option --opname is deprecated: support discontinued
    default: invalid options: -SHN
    default: (invalid options are ignored)

Not installing gem and therefore have no way of building and running cosmos in VM.

seanjone, 2019-06-04:
Master. Should I be using something else?

jlucas9, 2019-06-04:
Nope, that's correct.  A few more questions just so I can better debug:
* Can you confirm your on the head of master?  
* Have you attempted to provision multiple times and continue to receive this error?

I'll go ahead and see if I can kick off a build of master commit 4a9d9f2 and see what I get.
Worst case I'd recommend going with the OVA provided with the release [here](https://github.com/nasa/nos3/releases/tag/v1_03_01) in the meantime.

seanjone, 2019-06-04:
Yes I am on the head and I have attempted to provision twice. I received the error both times.  I will try out the OVA. Let me know if there is anything I may be doing incorrectly and thanks for the help.

seanjone, 2019-06-04:
Im not sure what changed but I did the provision again and it seems to have worked. Everything looks good now.

closed 2019-06-04 by seanjone