nasa/nos3 #21: VM nearly empty after installation

delazizi, 2019-06-04:
My VM opens directly to a terminal window and has no desktop interface. Its home directory contains two empty directories: nos3 and vagrant. There are no build scripts, no COSMOS... it's nearly empty!

Notes:
- I was never given the option to select the level of install, neither when I used vagrant nor when I used the windows installation script
- There were multiple issues when I ran vagrant up for the first time - most of them were 'No such file or directory' (in reference to many different paths)
- I have repeated the process 4 times. It is faster each time, but the resulting VM is identical. The issues I mentioned a moment ago have not been reproduced each subsequent time I run vagrant up.

jlucas9, 2019-06-04:
Hi delazizi,
I have a few minor questions for you:
* What branch and commit are you using?
* Have you completely destroyed (`vagrant destroy`) the VM between each run?

I'd recommend going with the OVA provided with the release [here](https://github.com/nasa/nos3/releases/tag/v1_03_01) in the meantime while I run some test builds as another issue with the COSMOS provisioning may be causing a bug.

delazizi, 2019-06-04:
- I am using master. Latest commit: 4a9d9f2
- I have not destroyed the VM at any point! Great to know! I'll destroy it and try again.

jlucas9, 2019-07-15:
Closing - let me know if you have any other issues!

closed 2019-07-15 by jlucas9