nasa/nos3 #24: Username and password for nos3 over vagrant

eyalbarlev, 2019-09-09:
Hello. Seems the vagrant username and password from your manual does not work . Any changes one needs to be aware of as for the username and password ?

jlucas9, 2019-09-10:
Hi eyalbarlev,  

Are you provisioning using vagrant or did you import the release 1.3.1 OVA?  

The user login / password should be: `nos3` / `nos3123!`  
The vagrant (root) login / password should be: `vagrant` / `vagrant`

If you don't see the typical Ubuntu login screen and instead see a back screen that would mean that either the provisioning failed or hasn't completed yet.  If you're doing the vagrant build I'd recommend updating to a release version if you aren't using one of those commits as well. The releases can be found [here](https://github.com/nasa/nos3/releases) with the associated documentation.  
  
I'll leave this issues open if you have any additional questions or issues. Have a great day!

eyalbarlev, 2019-09-12:
Hello , and thank you for your reponse.
I did not have a multitude of options, rather to download the latest vagrant version ( 2.2.x) and the latest vistualbox version (6.0) . the vagrant/vagrant password indeeed works , and yet , I am back to the main console with a prompt. nothing of nos3 comes up . This is a SOC with a 64Bits AMD and a GPU with ubuntu 16.04 . The virtualbox took indeed 20 minutes and updated everything inside to UBuntu 18.04 . After failing to start , the console window from which vagrant was executed prompted me to go for the vagrant provision , which I tried . I then started to see some green lines which indicated that things are going well, following a multitude of red lines with error, after which everything stopped . of course, I have all of your sources so that if you have a way around it , or you believe I am doing something wrong , please advise.
kind regards.

jlucas9, 2019-09-17:
Hello eyalbarlev,  
It sounds as though the vagrant provisioning process is having some issues.  The initial `vagrant up` command should open a black screen with a vagrant login, then you just have to wait until it installs everything from the desktop on up.  This can take anywhere from ~15 minutes to an hour and a half depending on your setup.  Once the `vagrant up` terminal exits you should be good to go and login to the NOS3 VM using the standard user and password. 

Occasionally, this provision process will fail due to various reasons and will need to be restarted.  I've included instructions do to this in the `README.md` file.  Also note that I have just added the OVAs to the [1.04.00 release](https://github.com/nasa/nos3/releases/tag/v1_04_00) so that you can simply download and import your VM.  

Please let me know if either of these don't work and have a great day!

closed 2019-11-14 by jlucas9