nasa/nos3 #27: cFS-to-AIT Interface Issues

EKlaseen, 2019-11-12:
Hello there,

I currently have a CentOS with AIT configuration set up on my host computer but I am running into some issues getting AIT to work properly. Per the instructions on the Nos3 webpage, whenever I initiate the run script a terminal window with two tabs and a google chrome window are supposed to appear for the ground station. However in my configuration, two terminals that are identical to one another with headers “nos3@localhost:~/AIT/AIT-CFS” appear and the text reads: 

_ValueError: insecure string pickle
/home/nos3/.virtualenvs/ait-cfs/bin/python: No module named virtualenvwrapper
virtualenvwrapper.sh: There was a problem running the initialization hooks.

If Python could not import the module virtualenvwrapper.hook_loader,
check that virtualenvwrapper has been installed for
VIRTUALENVWRAPPER_PYTHON=/home/nos3/.virtualenvs/ait-cfs/bin/python and that PATH is
set properly.
(ait-cfs)_

I did install virtualenv AND virtualenvwrapper prior to installing Nos3 so I’m not entirely sure why I am getting this error. Furthermore, as previously mentioned I am getting duplicate terminals for AIT-CFS in addition to duplicate terminals for the STF-1 flight software whenever I initiate the run script. I believe the issue lies somewhere between the cFS and AIT interaction, but I'm not entirely sure. 

I did install a prior version of Nos3 back in July with a COSMOS configuration and had similar issues of not being able to get COSMOS working properly. The user’s guide instructed me to select target “Nos3”and send command “Enable Telemetry” so that telemetry could start flowing into COSMOS and I could view it through packet viewer. But when I went to the command sender window, I could not find “Nos3” as a target. The only targets available were “Mission”, “System”, and “CFS” and none of them had an “Enable Telemetry” command. 

To sum up, it seems no matter what ground station I choose the cFS to ground station interaction is off-nominal. Not sure where to turn next so any guidance offered would be of great help. Thanks.

mgrubb-gsfc, 2019-11-12:
Hello, 

We will have to look into the AIT/Centos issue. For the COSMOS build, it sounds like the documentation is out of date. You can easily add new COSMOS targets to send the command to enable telemetry, but the easiest way would be to use the MISSION/CMD_UTIL function. The COSMOS telemetry input is connected to the TO_LAB application (MSG_ID 0x1880) and the command to enable the telemetry is TO_OUTPUT_ENABLE_CC (6). That should get the telemetry flowing. 

Thanks, 
Matt

mgrubb-gsfc, 2019-11-14:
With respect to the AIT on Centos configuration. The issue could potentially be in your ./bashrc file. See below from one of our developers:

"export WORKON_HOME=$HOME/.virtualenvs
export PROJECT_HOME=$HOME/Devel
source /usr/local/bin/virtualenvwrapper.sh
aren't mapped correctly.
in ubuntu virtualenvwrapper.sh is in usr/local/bin but in centos it's in /usr/bin"

Let us know if this helps

closed 2022-09-15 by jlucas9