nasa/nos3 #28: Documentation versus Implementation

PeteBlanchard, 2020-01-21:
Hello,

Less of a issue, more of questions...I am new to NOS3...I was able to clone and get an instance running with COSMOS, but:
*  the command sets are very different than what is shown in the documentation [User's Manual](http://www.stf1.com/NOS3Website/docs/NOS3_CombinedDocumentation.pdf).  e.g. there is no 'STF-1' target, but there is a MISSION (did it replace it?)
  * there is no ENABLE_TELEMETRY command
  * is the STF-1 command set still available (I didn't see anything in the commit history that showed it being removed...or maybe I missed it)?
  * the CMD_UTIL allows sending generic CCSDS command function codes
    * what are they?  are there generic/simulated sets for NAV, CAM (camera?), HK, etc.  commands?  or are they all per satellite/per mission?
    * using CMD_UTIL, can not send any additional parameters, won't most commands require parameters?
  * COSMOS output is very different than when running their docker container which constantly scrolls logging information about health and status from their simulated targets...
    * I would expect sending a 'MISSION CAM_HW_CHECK_CC' would result in a MISSION_TLM packet of CAM_HK_TLM_T, but nothing comes through...I also don't see anything in the NOS Engine tab of the terminal (not sure if I am supposed to)
* there aren't 8 simulators running, only 4:
  * GPS Simulator
  * CAM Simulator (Not listed in documentation)
  * Simulator Terminal
  * NOS Time Driver
  * missing: Magnetometer Simulator, EPS Simulator, Antenna Simulator, Cadet Simulator
    * are these no longer supported?
    * are they in a different repository?

Thanks in advance.

hstankey, 2020-03-30:
@PeteBlanchard I'm wondering the same thing. Did you find a solution to this?

PeteBlanchard, 2020-03-30:
@hstankey So far, I have received no response from any ticket that I've added to the system.  Not even a "we will evaluate your request" or "it is not in our scope at this time"...

hstankey, 2020-03-30:
@PeteBlanchard I've found this to be similar to your question: [Issue #8](https://github.com/nasa/nos3/issues/8)

> It appears that the documentation was developed using the Simulation To Flight - 1 (STF-1) version of NOS3 which includes mission specific simulators that have not been released.

PeteBlanchard, 2020-03-30:
@hstankey Good find!  I thought I had looked before submitting my question, but that IS similar...unfortunately, without that version of NOS3, it really doesn't do much.  The released interface doesn't do anything significant (e.g. turn the satellite, extend a solar panel, take a picture)

mgrubb-gsfc, 2020-03-30:
Hey @PeteBlanchard and @hstankey 
We are working on a new release of NOS3 that should be out in April. This will include a lot more functionality between 42, and simulated components, such as slewing the spacecraft. I will try and get our documentation updated to reflect the release.

PeteBlanchard, 2020-03-30:
Thanks for the update, that would be awesome.  I look forward to seeing.  Cheers!

hstankey, 2020-03-30:
@mgrubb-gsfc likewise, thanks for the update. Will definitely be trying it out when it rolls out.

atk5025, 2020-03-31:
@mgrubb-gsfc adding my interest in seeing the upcoming release. Looking forward to it!

mgrubb-gsfc, 2020-05-13:
Just to update this thread. If you would like to see the newest release, you can checkout rc-1.05.00 branch and follow the readme to generate the VM and build nos3. The readme's are up to date, but we are still working on the user's guide. This is a release candidate, and official tag is planned for July

PeteBlanchard, 2020-09-29:
has there been any progress on the user's guide?

mgrubb-gsfc, 2020-10-08:
The updated User's Guide is attached to the release 1.05 on the releases tab, and the website nos3.org should also be updated for the new release. Let us know if there is something there you are looking for that we missed.

DeepCodeSec, 2020-11-05:
For those reading the manual lined aobe, note that key sections of the manual above are obsolete, namely section 5.4. As per issue #43, to enable telemetry in NOS3 as of November 2020, in the COSMOS "_Command Sender_" window, use the following values:

1. Select “Target:” to be “**CFS**”
2. The only “Command:” is “**TO_ENABLE_OUTPUT_CC**”

Then use the _Packet Viewer_ to start reading telemetry.

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

closed 2022-09-15 by jlucas9