nasa/nos3 #29: __pthread_mutex_lock_full

cappelaere, 2020-01-23:
It happens fairly often after quitting the application and restarting it.
In terminal window:
...
EVS Port1 42/1/CFE_TIME 1: cFE TIME Initialized
2000-012-14:03:20.20220 ES Startup: Core App: CFE_TBL created. App ID: 4
EVS Port1 42/1/CFE_TBL 1: cFE TBL Initialized.  cFE Version 6.5.0.0
2000-012-14:03:20.25304 ES Startup: Finished ES CreateObject table entries.
2000-012-14:03:20.25340 ES Startup: CFE_ES_Main entering CORE_READY state
EVS Port1 42/1/CFE_EVS 1: HWLIB Initialized. Version 1.0.0.0
EVS Port1 42/1/CFE_EVS 1: CAM Lib HW Init Success
NAV_LibInit(): Initializing the GPS
core-linux: ../nptl/pthread_mutex_lock.c:422: __pthread_mutex_lock_full: Assertion `INTERNAL_SYSCALL_ERRNO (e, __err) != EDEADLK || (kind != PTHREAD_MUTEX_ERRORCHECK_NP && kind != PTHREAD_MUTEX_RECURSIVE_NP)' failed.
bash: line 1:  2625 Aborted                 (core dumped) ~/Desktop/nos3-build/linux/linux/core-linux

besides rebuilding, I have not found another way of being able to restart...
I am suspecting that one of the threads does not exit cleanly.
Thanks.
Pat.

cmanderino, 2020-01-30:
I've seen this happen with linux-based threads that make SYS calls outside the OSAL. IIRC sometimes a PR reset fixed the issue, but one thing that helps a lot of cFS troubles is to not run cFS with sudo and make sure you are running with extended mqueues instead.

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

closed 2022-09-15 by jlucas9