nasa/nos3 #31: 42 and NOS3

PeteBlanchard, 2020-02-03:
I apologize for asking such a n00b configuration question, but I was recently looking at the NOS3 and wanted to be able to run multiple NOS FSW's/Engines/Apps (to simulate interacting with multiple satellites)...the 42 implementation (that NOS3 uses) seems to tie me to a single instance of the NOS Engine configured through the Inp_NOS3.txt (which seems an extended implementation of the original time driver for 42).  I read that as being a NOS Engine per satellite, but since 42 can simulate multiple satellites...I was looking for a way to configure/run multiple...

I checked out the latest version of 42 to see what was there, but other than the header stubs, it seems to only support using NOS3 time (and the send/receive methods in the commit you use are no longer there; except in a header file, no implementation).  Mr. Stoneking said that he re-worked how 42 manages and uses its sockets (and pointed me at his 'Standalone' demo), but that ACS implementation listened for inputs during startup and it seemed to me that 42 used to connect to the NOS Engine 'bus' and send/receive data that way...he did add a new ACS and TXRX mode to his Inp_IPC.txt configuration...the ACS seems close, but his example shows a SERVER mode with an AcApp...not sure if I could just change it to CLIENT and point it at the NOS3 engine...

Has there been any work towards an implementation like this?  Or am I looking at it wrong?

Thanks in advance.

mgrubb-gsfc, 2020-03-30:
We have the new version of 42 included in our upcoming release. It does use the Inp_IPC file to specify each of the sockets that a simulator will connect to. One of the connections is the NOS time interface, but the others connect directly to each different simulator. 

We use NOS3 for development of a single spacecraft FSW and ACS code, so I have not given much thought to a multi spacecraft. I would personally run multiple instances of NOS3 VMs for each of the individual spacecraft. 

@cmanderino do you have any inputs here on how you planned to do multi-mission NOS3.

PeteBlanchard, 2020-03-30:
Great, that (first part about 42 interfacing) sounds good.

I was viewing the COSMOS as (essentially) a ground station (e.g. talks to multiple systems on multiple sats).  The 42 software (itself) supports multiple satellites (all across the solr system, see demo).  It seemed reasonable (to me) to be able to run multiple NOS3 (STF-1) in a single 42 with COSMOS communicating between.

PeteBlanchard, 2020-06-03:
@mgrubb-gsfc is that in the rc-1.05.00 branch?

timothy, 2020-07-13:
I would like to have more than one satellite running in my NOS3 setup as well. It would be very cool if there was a way to load full constellations of sats. If anyone has documentation that I can peruse it would be much appreciated!

msuder, 2020-07-14:
Yes, this is doable, but as mgrubb-gsfc said, we don't have any examples as of now.  Here are the things you would need to look at to make it work:
1.  Specify multiple satellites in the 42 input files (specifically Inp_Sim.txt):  See https://github.com/ericstoneking/42/blob/master/Docs/42%20Overview.pdf:  Slide 28 starts the description on the input files and slide 29 actually shows multiple reference orbits and spacecraft being defined.
2.  Specify multiple "prefixes" (e.g. SC[0], SC[1], ...) in the 42 input file Inp_IPC.txt to send data for over multiple sockets.
3.  Write your sim to parse the data for a specific "prefix".  See for example https://github.com/nasa-itc/novatel_oem615_sim/blob/40dc91b857bd089af1053055861b37c1e4ea0db9/src/gps_sim_data_point.cpp, method GPSSimDataPoint::do_parsing and note that the variable "MatchString" is created to match a specific "SC[N]" prefix, where "N" is specified (indirectly by parsing in gps_sim_data_42socket_provider.cpp and it setting the GPSSimDataPoint _sc member variable) in the sim configuration file (usually named "nos3-simulator.xml").
4.  Note that 42 does not pass the NORAD S/C ID or anything like that, so the numbering of the spacecraft is just an index based on the ordering in the 42 config file... so you need to make sure that you match up the right spacecraft number in 42 with the spacecraft number you want in the sim.

timothy, 2020-07-23:
Thanks for the response @msuder,
After looking over your recommendation and doing a simple proof of concept I came up with some additional questions.  

**Below is what I did:**
1.	Inside of the `Inp_Sime.txt` file I added the below:
```
**************************  Reference Orbits  **************************
2                               !  Number of Reference Orbits
TRUE   Orb_LEO.txt              !  Input file name for Orb 0
TRUE   Orb_LEO.txt              !  Input file name for Orb 1
*****************************  Spacecraft  *****************************
2                               !  Number of Spacecraft
TRUE  0 SC_NOS3.txt             !  Existence, RefOrb, Input file for SC 0
TRUE  1 SC_NOS3.txt             !  Existence, RefOrb, Input file for SC 1
```
2.	At this point when I load NOS3 I see two satellites in the 42 GUI. 
3.	As a proof of concept I went in and reconfigured the `gps_sim_data_point.cpp` `do_parsing` method to parse both satellite 0 and 1. I output the data of the extra sat to the terminal. I did not add anything to `nos3-simulator.xml` (just hard coded values in the .cpp for proof of concept).   

**Questions:**

1. Can NOS3 handle simulating multiple satellites out of the box? Is there a way to configure NOS3 to support multiple satellites without going through and editing the code and manually building the functionality into NOS3?  

1. Would it be a better approach to run hardware sims for each new satellite instead of configuring each sim to handle more than one satellite? For instance, instead of gps sim parsing both satellite and simulating both - would it be better to have one gps sim for each satellite? A one for one match with satellites and simulators?

1. Will I have to reconfigure each of the NOS3 simulators to handle more than one satellite?

1. Does this mean I will need to find a way to synchronize them all to the same time driver, configure each to talk to a single ground station, etc?

cmanderino, 2020-07-23:
@timothy 

Hi, just chiming in, here. I've run a constellation of satellites in 42 with cFS instances a number of times over the past couple years and I wanted to throw in just a quick reply to your questions:

1. Which NOS3 version are you working with and how are you launching it? 
1a. There's no NOS3 script that I'm aware of that does this.
1b. Editing which code? 

2. I would recommend running hardware sims for each satellite. Remember, each sim and flight software (cFS) instance is intended to run a single spacecraft. You wouldn't need to only duplicated the sim, you'd have to rewrite every ADCS application that uses it to understand that it is running more than one spacecraft. This isn't really scalable, and while it might be a fun coding exercise I don't believe that it will offer what you want.

3. So, NOS3 is comprised of the 42 Spacecraft Simulator, COSMOS Ground Station, cFS Flight Software, and Spacecraft Hardware Simulations. If you want to send commands from COSMOS to a cFS instance, you need to either have multiple ground stations or multiple command lists - one set for each satellite flight software you are communicating with.

4. If you want realistic timing, yes. RE: COSMOS, see above. 

I think there is a paper coming out that uses this particular set-up that you might be looking at coming from University of Pittsburgh, I'll try to keep you tuned.

msuder, 2020-07-24:
@timothy A few follow-ons to @cmanderino comments:

1.  Concur with @cmanderino - It's not there out of the box, but is a simple matter of changing the 42 input files (Inp_Sim.txt), etc. as you have done - this gets multiple sats in 42.

2.  Concur with @cmanderino - The most expedient way to do this would be to replicate the <simulator> block in nos3-simulator.xml that contains <name>gps</name> and change the name (e.g. "gps2")... then use the same code in novatel_oem615_sim, but make another version of gps_simulator.cpp where you change std::string simulator_name = "gps"; to reference your new name (e.g.  std::string simulator_name = "gps2";) - This is kind of hacky and long term I'd like to have a better solution, but it would work for now for what you want.

3.  Concur with @cmanderino with slight tweak:  I would probably create multiple COSMOS interfaces (which will equate to multiple UDP socket pairs between COSMOS and cFS), one per satellite.  After that, you could run one or multiple instances of COSMOS depending on what you want to do (i.e. act like 1 operator controlling multiple satellites or multiple operators each controlling their own satellite).

4.  The way NOS3 works, the timing between 42, all of the sims and flight software will always stay in sync... the only thing that is "free wheeling" separately is COSMOS.  So it depends on whether you need COSMOS time synchronized to the other  pieces or not.

@cmanderino I would be interested in the paper when it comes out also.

timothy, 2020-07-25:
Thank you @cmanderino and @msuder for your responses.   

**Questions**: 

1. Does it make sense to have each satellite run on a different hardware bus (NOS engine)? 
2.  Will it scale well if all the satellites run on the same hardware bus?
* I ask because it sounds like both satellites will share the same NOS engine If I change the `simulator_name` variable to hold "gps2" inside the new `gps_simulator.cpp` and then go into the `nos3-simulator.xml` and add a new sim for my new GPS2. 

@cmanderino  I am interested in reading the paper as well, please share it when it is available ☺

msuder, 2020-07-27:
Sorry, yes, you should also specify different uart busses (assuming UART bus) so that the virtual connections between the 2 CPU/C&DH and the 2 GPS are different... otherwise there will be crosstalk on the bus.

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

closed 2022-09-15 by jlucas9