nasa/nos3 #32: FSW not connected to port upon startup.

hstankey, 2020-03-27:
I'm getting the following error on the NOS Engine Standalone Server terminal after building and running NOS3 on the VM.

> [ ERROR ] NosEngine.Uart - close uart port failed: node (fsw) not connected to port (1)

It's also accompanied with the following error on the NOS3 Flight Software terminal.

> Error, Can't Open ES App Startup file: /cf/cfe_es_startup.scr EC = 0xFFFFFFFF

Has anyone seen this or know what I may be missing?

mgrubb-gsfc, 2020-03-30:
The NosEngine.Uart message is typical. Nos Engine allows reconnecting of components that have disconnected, so it attempts to close the port before connecting. On the initial launch, the close will fail because nothing had connected yet. 

The second error may be because of statically linking. I believe the current release skips the startup.scr. If you see the SCH load and still see that error then everything is working properly. That issue will be resolved in the next release.

PeteBlanchard, 2020-06-04:
I have seen that same message, but the flight software appears to be running (just confirming that someone else saw the same messages, but as you point out the cFS appears to be working).

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

closed 2022-09-15 by jlucas9