nasa/nos3 #33: nos3 login

goalomy, 2020-04-21:
![VirtualBox_NOS3_Ubuntu_21_04_2020_16_58_34](https://user-images.githubusercontent.com/63917677/79861156-8df19c00-83f1-11ea-84d9-be42d3918a46.p

unable to enter nos3 login and password.What i need to do please suggest the way.

Abhishek5101, 2020-05-02:
The root username is vagrant and password is vagrant

PeteBlanchard, 2020-06-04:
It appears that your build may have run into difficulty...

I am not an expert on it, but when I do my vagrant builds, if I am behind a corporate firewall that re-writes certificates, I have to update the local certificate store with the company 'trusted' cert authority...I also export a couple environment variables for other tools that use ssl to work properly (in the /etc/bash.bashrc)
```bash
export CURL_CA_BUNDLE=/etc/ssl/certs
export SSL_CERT_FILE=/etc/ssl/certs/ca-certificates.crt
```

After you do that, you may need to re-run the build scripts

closed 2022-09-15 by jlucas9