nasa/nos3 #36: Reaction Wheels

rcooper64, 2020-06-03:
I have been delving into the code to figure out how to add in 2 more reaction wheels to stabilize all axis of rotation. I have been successful in adding in two new reaction wheels into the command center, but i cannot find where all the communications go. Also, how can I change the axis that the wheel rotates at?

rcooper64, 2020-06-04:
I have found a way to duplicate the reaction wheels. am just looking for rotating the axis of the individual wheels

msuder, 2020-06-04:
The configuration of the number, position, and other parameters for the reaction wheels is done in the text files that configure 42, the dynamic spacecraft simulator that is incorporated into NOS3.

The configuration files end up in /opt/42/NOS3InOut in the latest rc-1.05.00 branch and the file you would configure is SC_NOS3.txt.  Look for "Wheel Parameters".  Each wheel has "Wheel Axis Components, [X, Y, Z]".

rcooper64, 2020-06-04:
Yeah I have seen that file. I am talking about the "generic reaction wheel" in the command sender that you can do live while running it. I have duplicated that to another wheel that is can be sent commands, but it is still on the same axis. Any ideas??

msuder, 2020-06-04:
When you duplicate the code, you will need to alter the method "GenericRWHardwareModel::handle_command" in generic_rw_hardware_model.cpp to specify a string different than "SC[0].AC.Whl[0].Tcmd = "... in particular, you want to change the second "0" to a different wheel number, corresponding to a different wheel in the 42 input file.  With respect to reading the wheel momentum, you need pay attention to the "GenericRWDataPoint::do_parsing" method in generic_rw_data_point.cpp file; although that one is better parameterized to use the wheel number from the nos3-simulator.xml config file.

rcooper64, 2020-06-05:
Okay so I swapped the second and third wheels to "SC[0].AC.Whl[1].Tcmd" and "SC[0].AC.Whl[2].Tcmd", but when I run the command on the sender, It still rotates only on the X axis... in the code somewhere would it still be sending the messages through the port that goes to the first wheel??

msuder, 2020-06-05:
There is a config file called "nos3-simulator.xml".. with a block called "generic-reaction-wheel" sim.  I'm not quite sure how you did the duplication, but you might need to make changes in that to get everything working.  Wireshark is useful for "seeing" the data sent over the sockets, such as between 42 and the sim.  "nc" is another useful tool for showing what is received on one end of a socket.

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.