nasa/nos3 #38: GUI Crashes on nos3 user Login

kialio, 2020-06-15:
Hi,

We are trying to get this up and running for development and have experienced immediate crashes of the VM when we try to login with the 'nos3' user in the X-GUI.  We can log into the VM via the command line ('vagrant ssh') and access the nos3 account that way.  

Note that if we switch users to 'vagrant' in the GUI we can log in without any issues.

Jeremy

kialio, 2020-06-15:
I can provide logs if needed.  We are seeing this happen on two different systems.

DKlaren, 2020-06-22:
I had a similar issue was able to get around it by changing the display adapter setting in virtualbox from VBoxVGA to VBoxSVGA.

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

closed 2022-09-15 by jlucas9