nasa/nos3 #43: Attitude Control

timothy, 2020-08-13:
I had a few questions on attitude. 

- Is there a pre-existing way to adjust the attitude of the NOS3 satellite? 
- If so how can I make the adjustment?
- If the satellite is pointing at the ground is there a way to determine the lat/lon values for the footprint on the ground?

msuder, 2020-08-13:
Hopefully this helps and stops short of being overwhelming.  The short answers are:
1.  Yes (see ***1*** below)
2.  In the 42 dynamics engine config files for the initial conditions and via reaction wheel torque commands as the scenario progresses in time (see ***2*** below)
3.  Kind of.  42 knows where the satellite is, and can represent a GPS sensor that can provide the lat/lon/alt of the satellite... it would be up to you to use that to determine a footprint on the ground (for a nadir pointer, the lat/lon provided would also be the subsatellite point lat/lon, but you may want to compute a "circular" or other footprint if you have a sensor with a certain field of view).  Also, it doesn't have to be a nadir pointer, your GN&C algorithms in flight software could take in position data (e.g. from a GPS) and attitude data (e.g. from gyros, star trackers, sun sensors, etc., etc.) and "imaging" sensor parameters (boresight axis, field of view, etc., etc.) and actually compute the sensor footprint on the ground.  (See ***3*** below)

Detailed answer (sorry, this got pretty long, but the real hardware answer is somewhat involved, so the FSW+Sim+C&T answer gets pretty long also):

There are a number of pre-existing sample applications, simulations, and ground command and telemetry databases to help you to see how attitude can be determined and controlled and how orbit can be determined from sample on-board sensors.  The two I'd like to call out that relate to your question are:
- Generic Reaction Wheel app (.../fsw/components/generic_reaction_wheel), sim (.../sims/generic_reaction_wheel_sim), and command and telemetry tables (.../gsw/cosmos/config/targets/GENERIC_REACTION_WHEEL)
- NovAtel OEM615 GPS app (.../fsw/components/novatel_oem615), sim (.../sims/novatel_oem615_sim), and command and telemetry tables (.../gsw/cosmos/config/target/NOVATEL_OEM615)

The generic reaction wheel C&T provides the ability to request data on current momentum telemetry and to send torque commands for a reaction wheel (specified in the 42 config)
The NovAtel OEM615 C&T provides the ability to request data from a GPS (specified in the 42 config) for time, position, and velocity

If you follow the user's guide instructions and run "make" and "make launch" you will get a number of windows... I've rearranged them a little, but they look like this:
[image]

The UI of the 42 dynamics camera window is on the left; the C&T server, command sender, and telemetry packet viewer are in the center; and the flight software and the simulators are on the right.  To get any telemetry, you have to send the "CFS" "TO_ENABLE_OUTPUT_CC" command.  If I then send the "NOVATEL_OEM615" "NOVATEL_GPS_REQ_DATA_CC" command, my screens will look like this:
[image]
So you will see that the flight software received a command to turn on output and that the packet viewer shows GPS telemetry (the time, position, velocity cannot be seen due to scrolling, here is a shot of them: 
[image]
)

So the command was sent from COSMOS to the FSW, the FSW "talked to" the GPS sim across a virtual UART connection between the FSW on the C&DH system and the GPS sim representing the NovAtel OEM615, and then it retrieved data from a TCP socket connection to the 42 dynamics engine that provided the time, position, velocity.  The GPS sim provided PVT telemetry back over the virtual UART to the FSW, and the FSW then "transmitted" it to ground in its telemetry stream.  So one thing to emphasize is the role of the 42 dynamics engine in this.  There are config files for it that specify all of the initial conditions for the satellite, including the epoch (date/time) for the initial conditions, the initial orbit, the initial attitude and attitude rates, what sensors (e.g. GPS) and actuators (e.g. reaction wheel) are on board, etc., etc.  The 42 config files are at .../sims/cfg/InOut in the git repo, but they get copied to /opt/42/NOS3InOut for execution of the dynamics engine.

***1*** The Inp_Sim.txt 42 file lists the "Input file for SC 0" on line 13 (default is "SC_NOS3.txt")... if you look in the SC file, lines 13-16 provide initial attitude and attitude rate conditions for the spacecraft, so here is where you would set those initial conditions.
***1*** In the default configs, the angular velocities are initially 0, so the spacecraft is not spinning.  You can send the "GENERIC_REACTION_WHEEL" "GENERIC_RW_SET_TORQUE_CC" command with a non-0 torque parameter (e.g. 1) and it will end up getting relayed from ground->fsw->sim->42 to apply a torque to the reaction wheel... in the 42 camera view, you will see the spacecraft begin to spin.  If you then send a "GENERIC_REACTION_WHEEL" "GENERIC_RW_REQ_DATA_CC" command, you can see the reaction wheel telemetry and not that the reported momentum (MOMENTUM_NMS) of the wheel becomes non-0.  Here is a screen shot:
[image]

***2*** How to configure the initial attitude/attitude rates was given in ***1** above, but there are a couple of other things to know/configure for the dynamics... the SC file also lists the spacecraft mass, moments of inertia, center of mass, etc. on lines 30-34.  And the definition of the reaction wheel (only one in this configuration) parameters is given in the default SC file on lines 62-68... parameters like the initial momentum, wheel axis, wheel rotor inertia, etc.

***3*** The GPS in the example also has parameter definitions in the SC file... in the default file they are on lines 191-195.

[... cut ...]
So part of the development of a system using NOS3 is creating sim models of your specific hardware components, including the byte level communication across the bus (UART, I2C, SPI, ...) to the C&DH and also any nuances of how the hardware works (e.g. for the GPS, 42 provides the time, position, velocity, but the OEM615 supports certain commands to request data, to request data be streamed at a certain rate, etc., etc.).  A last comment related to the OEM615:  we only modeled the amount of functionality in the sim that we needed (and you should also)... the OEM615 supports a whole host of UART commands that we didn't implement because we didn't need them for what we were doing.

PeteBlanchard, 2020-08-13:
That is great information, thanks.  I was working on something similar (more along the lines of his 3rd question; down look of where a camera or other sensor could "see") and had come across some of that myself (specifically the OEM615 support) and the sample commands/telemetry.  

I had thought about expanding the GPS Sim to process the 'bestposa' as well, just so I didn't have to do the conversions myself...then there would need to be a new cFS app and Sim (attitude control system) to support getting the "Quaternion of S/C (B) in ECI (N)" (SC[0].B[0].qn) to get what (if any) rotation the satellite was under to be able to affect the field of view...then apply sensor aperture size to distance and create a bounding box (or something)...

Speaking of FOV, I notice that the 42 Camera has an FOV layer, but when I turn it on the drawings seem to originate somewhere (way) off the satellite (which I thought was odd).  When I looked at the Inp_FOV.txt, I noticed that where you specify the "Position on body (m)", for the "SOLID" to draw from is set to "0.0 0.0 1.0" which seems odd for a 10cm x 10cm x 34cm (3U) CubeSat.
[image]

So, I tried updating it to be "0.0 0.0 0.05" (actually set all of the drawings to use 0.05 in place of 1.0) and while that drew the origin point closer together...they are still nowhere near the satellite.  What am I doing wrong with that (other than not taking into account the 34 cm "height")?  I didn't notice an offset for formation flying, but maybe I missed something else (though wouldn't 42 have taken that into account?).
[image]

msuder, 2020-08-13:
First a reference:  https://github.com/ericstoneking/42

For all things 42, that’s the place to go. The main README page has references to the docs which will go in to much more detail than I did above. 

That said and without looking in detail at the config files right now, it’s probably something small in the config files and understanding the parameters.  We have not done much with the FOVs in NOS3, so the defaults for FOVs are probably not very reasonable (your observation on the 1.0 meter offset is a good example of that).  The origin of the model and what its axes are is going to matter also, so you might check that (42 recommends Wings3D as one tool to view/convert models).  Iirc, it’s just the model and the FOV parameters that are going to determine the relationship between where the FOV is and where the model is (and in turn where they are rendered visually in the camera view).  Iirc, FOVs are tied to spacecraft, not reference orbits, so the formation flying offsets should have no effect.

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

closed 2022-09-15 by jlucas9