nasa/nos3 #44: Writing sample_app to push MIL-STD-1553 commanding to NOS Engine

sunghhong, 2020-08-19:
Hello, 

I am trying to modify a sample app (https://github.com/nasa/sample_app) to push 1553 data traffics to NOS engine.  I am eventually trying to have another external app connect to NOS engine to communicate 1553 data exchange.  Any idea what API can be used to support this effort?

The following locations did not have enough information to support 1553 protocols. 
https://github.com/nasa-itc/hwlib/tree/9e6618736e4265e6f951c8224ec171a50a1873ae

But, NOS engine user guide (http://www.stf1.com/NOS3Website/docs/20170120_TESS_TO9_NOSEngineUsersGuide_v01.pdf) mentions 1553 plug-in in Section 5.  Any idea where I can find its API in the NOS3 github?

Thank you!

DVCoder, 2021-01-12:
Any update on this issue?

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

Will add it to the short list to provide all HWLIB interface examples in the sample app!

closed 2022-09-15 by jlucas9