nasa/nos3 #46: Unit Testing

PeteBlanchard, 2020-09-23:
Hello, I was working on trying to create my own subsystem and wanted to add in unit tests...I see several cFS components define unit test in the "unit_test" folder, but they don't seem to get built or run during the normal 'make' (and there is no 'test' target).

Also, I ran into the following issues:
* missing some of the referenced include files...things like "stf1_test.h", "i2c_hooks.h"
* was trying to pull in the "ut-assert" from some of the 'apps' (to, ci, cf, iolib), but ran into an issue with them re-defining functions from the 'ccsds.h' (apparently not 'ifndef' protected)...would be nice if they were in a central library that could be "included" instead of copied into each 'app' or 'component'

PeteBlanchard, 2020-09-29:
Also tried using the following (from "cfe/cmake/arch_build.cmake"):

* add_unit_test_lib
* add_unit_test_exe

but the build fails with:

```bash
CMake Error at /home/nos3/nos3_dev/fsw/cfe/cmake/arch_build.cmake:194 (add_executable):
  Cannot find source file:

    /src/utexec.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
```

And I cannot find that file in the source anywhere either...

mgrubb-gsfc, 2020-10-08:
@PeteBlanchard The unit test framework is currently broken in cFS version 6.7. We did the same things you tried to get it working and decided to hold off until the official cFS 6.8 is released (hopefully very soon). I will update NOS3 to 6.8 and the cFS Sample app will provide a proper unit testing example.

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

Updates to even the newer version (7.2) of cFS planned and unit test fixes planned!

closed 2022-09-15 by jlucas9