nasa/nos3 #53: Changing the satellite position dynamically through COSMOS

jayaramanjay97, 2020-11-22:
Is it possible to dynamically change the satellite's position through COSMOS.
Like the Generic wheel simulator can change torque and the direction the satellite rotates, is it possible to change the parameters of satellite from COSMOS?
e.g. To change the altitude of the satellite by sending a command from COSMOS?
Any help would mean a lot.
Thanks
Jay

mgrubb-gsfc, 2020-11-25:
Jay, 

This is not possible in the current configuration. That would need a thruster added to the satellite model to produce the change in altitude. The functionality of the dynamics simulation is there, but the simulator for the thruster and an application to control it would need written.

You can however change the orbit parameters for the entire simulation to change the altitude when it starts. This can be done through the files in sims/cfg/InOut/Orb_LEO.txt

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

closed 2022-09-15 by jlucas9