nasa/nos3 #54: user guide.

shaileshdeshpande, 2020-11-24:
Hello, 

I am a new user for the NOS3 simulation. I am trying to get the code and run locally. I am using the virtual box and following the instructions.  Step 1a) is not coming in the new virtual box. Is there a new version of the instructions manual

1) Access the nos3 repository on the Host Machine. From the Guest:
**a) Use Virtual Box – Guest Additions CD Add**
b) Virtual box Menu > Devices > Shared Folder > Shared Folder Settings
c) Add new Shared Folder and select the location of nos3 repo on host
d) Check Select Auto Mount, Make Permanent
e) Reboot VM
2) Make and Install software
a) Navigate to the nos3 shared folder after VM Reboot from Step 1e
b) Navigate to the nos3 shared folder and nos3 parent directory
c) Run “make”

mgrubb-gsfc, 2020-11-25:
This is a step to install the virtual box tools to add shared folders. See the link here for the manual https://www.virtualbox.org/manual/ch04.html

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

closed 2022-09-15 by jlucas9