nasa/nos3 #55: Running NOS3 instruction 5.3

shaileshdeshpande, 2020-11-24:
The command make launch is failing at line 83. I have given access to execute, but it is still not working.

mgrubb-gsfc, 2020-11-25:
Can you post the error you are seeing? The launch script is only 48 lines long, so I am unsure what is showing an error on line 83.

W-E-V-S, 2020-12-29:
Hello, I'm getting the same. Error below:

nos3@nos3:~/Desktop/NOS_Git_Repo/nos3$ make launch
./gsw/scripts/launch.sh
: invalid option
Usage:	/bin/bash [GNU long option] [option] ...
	/bin/bash [GNU long option] [option] script-file ...
GNU long options:
	--debug
	--debugger
	--dump-po-strings
	--dump-strings
	--help
	--init-file
	--login
	--noediting
	--noprofile
	--norc
	--posix
	--rcfile
	--restricted
	--verbose
	--version
Shell options:
	-ilrsD or -c command or -O shopt_option		(invocation only)
	-abefhkmnptuvxBCHP or -o option
Makefile:83: recipe for target 'launch' failed
make: *** [launch] Error 2
nos3@nos3:~/Desktop/NOS_Git_Repo/nos3$

W-E-V-S, 2020-12-29:
Hello, same here:

- nos3@nos3:~/Desktop/NOS_Git_Repo/nos3$ make launch
- ./gsw/scripts/launch.sh
- : invalid option
- Usage:	/bin/bash [GNU long option] [option] ...
- 	/bin/bash [GNU long option] [option] script-file ...
- GNU long options:
- 	--debug
- 	--debugger
- 	--dump-po-strings
- 	--dump-strings
- 	--help
- 	--init-file
- 	--login
- 	--noediting
- 	--noprofile
- 	--norc
- 	--posix
- 	--rcfile
- 	--restricted
- 	--verbose
- 	--version
- Shell options:
- 	-ilrsD or -c command or -O shopt_option		(invocation only)
- 	-abefhkmnptuvxBCHP or -o option
- Makefile:83: recipe for target 'launch' failed
- make: *** [launch] Error 2
- nos3@nos3:~/Desktop/NOS_Git_Repo/nos3$

PabloSoligo2014, 2021-02-04:
I had the same problem when I used windows 10 as host. Now I test it with ubuntu as host and it works fine...at least up to this point..

Derrekito, 2021-08-11:
The issue stems from Windows EOL incompatibility. If you edit a file in windows or clone the repository using an incorrectly configured git-bash - then don't be surprised to see this. Here a fix for you:

`#!/bin/bash

# Convert EOL from DOS to UNIX 
#
# arg is the direct path to the broken file
# USAGE EXAMPLE: ./fix_eol.sh /nos3/gsw/scripts/launch.sh

BROKEN_FILE="$1"

if [ $# -eq 0 ]; then
    echo "ERROR: arg missing"
    echo "USAGE EXAMPLE: fix_eol.sh /nos3/gsw/scripts/launch.sh"
    exit 1
fi

sed -i.bak 's/\r$//' $BROKEN_FILE`

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

closed 2022-09-15 by jlucas9