nasa/nos3 #56: cant login with nos3 user

shahin93hashemi, 2020-12-07:
### **hi all
**I installed the program according to the instructions below.**

----------------------------------------------------------------------------------------------

Installing
1  Open a terminal
2   Navigate to the desired location for the repository
3   Clone the repository git clone https://github.com/nasa/nos3.git
4   Clone the submodules git submodule init and git submodule update
5   Navigate to /nos3/deployment
6   Run vagrant up and wait to return to a prompt
-This can take anywhere from 20 minutes to hours depending on internet speeds and host PC specs
-The VM will reboot multiple times in order to finish install packages for you automatically so wait for that prompt!
7   Login to the nos3 user using the password nos3123! and get to work!
8   Try building and running following the instructions below

--------------------------------------------------------------------------------------------

**Everything went well until the sixth stage of installation
But
After Run command "vagrant up" and displaying the login page. The nos3 user is not displayed and I have to log in only with the vagrant user.
i uploaded pictures from vm.
how i can login with nos3 user?
thank you**

[image]

[image]

efriesema, 2020-12-24:
I'm in a similar place    command prompt where I ran vagrant up   it displayed the following so itlooks like something was wrong with ansible loading but there is no way to preload Ansible onto a windows machine
Windows version 10
VirtualBox  6.1 
Git 2.21.0.windows.1
Vagrant 2.2.14
==> dev: Mounting shared folders...
    dev: /vagrant => C:/Users/Ed/Source/Repos/NOS3/nos3/deployment
    dev: /tmp/filestore => C:/Users/Ed/Source/Repos/NOS3/nos3/deployment/nos3_filestore
==> dev: Running provisioner: ansible_local...
    dev: Installing Ansible...
The following SSH command responded with a non-zero exit status.
Vagrant assumes that this means the command failed!

                add-apt-repository ppa:ansible/ansible -y &&                 apt-get update -y -qq &&                 DEBIAN_FRONTEND=noninteractive apt-get install -y -qq ansible --option "Dpkg::Options::=--force-confold"

Stdout from the command:

Reading package lists...

Stderr from the command:

E: Could not get lock /var/lib/apt/lists/lock - open (11: Resource temporarily unavailable)
E: Unable to lock directory /var/lib/apt/lists/

hailhallmonitor, 2021-07-23:
I am attempting to get past the vagrant login as well.

satzguyz, 2022-05-21:
The problem is incomplete installation. Try to destroy the vagrant and re-install again.

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.

closed 2022-09-15 by jlucas9