nasa/nos3 #73: nos3 Uncompleted Installation

M-i2cat, 2021-11-15:
Good Morning,

First of all thanks beforehand for the help. 

Secondly, I'm having some trouble launching and completing the installation of nos3 (Version nos3_dev_1.05.0). The virtual machine opens up correctly and I can log in to the nos3 user correctly. 
Once there it seems that I'm missing some steps as I can't progress.

I have done different approaches:

1. Following the Users_Guide_v1.04 I get stuck in this step:
 5.2 Building NOS3
 Log in to the NOS3 VM: nos3 / nos3123!
 1- Double click “nos3-build.sh”
As it seems that the file nos3-build.sh does not exist in the VM. (I have also observed that the installation guide seems to use a different configured virtual machine using Ubuntu than the one in this last version )

2. Following the readme file. Once I share the nos3 folder from windows to the VM again the forces seem to fail when trying to run the make commands
"
It is recommended to share the nos3 repository into the virtual machine
1- Open a terminal
2- To build use the `make` command from the nos3 repo
3- To run nos3 use the `make launch` command from the nos3 repo
4- To halt nos3 use the `make stop` command from the nos3 repo
"
I have cleaned vagrant and the VM environment to discard possible errors during the installation but the errors persist. 

Hoping that you could give me a hand in solving this process. 

Thanks a lot

hende101, 2022-02-28:
I also have this issue, was this ever resolved?

jlucas9, 2022-09-15:
Closing issue due to age.
Please create a new issue if this problem remains.
User guide update in progress - will be moving to use GitHub Wiki!

closed 2022-09-15 by jlucas9