nasa/nos3 #78: Docker Build

jlucas9, 2022-08-26:
Add an option to deploy and run using docker.

jlucas9, 2022-10-20:
I got to:
* Manually installed docker on my VM
* Created a docker container using the https://github.com/nasa-itc/deployment/commit/f4ca12f4e67db9a5a8a5b97e90cf4964c32d97ea
* Created a `docker_launch.sh` similar to the regular launch 
  * Am able to build all parts manually
  * Am NOT able to run the standalone server:

```
echo "Docker NOS Engine Server..."
cd $SIM_BIN
gnome-terminal --tab --title="NOS Engine Standalone Server" -- /usr/bin/nos_engine_server_standalone -f $SIM_BIN/nos_engine_server_config.json

    docker run -dp 12001:12001 -v $SIM_BIN:/build nos3 /usr/bin/nos_engine_server_standalone -f /build/nos_engine_server_config.json
```

msuder, 2022-10-27:
Notes from previous meeting "NOS3 Docker Tag Up"  onenote:https://nasa.sharepoint.com/teams/JSTARSmallSatelliteSupport/SiteAssets/JSTAR-%20Small%20Satellite%20Support%20Notebook/DSM.one#Meetings&section-id=%7B29B3AA95-B06A-41B8-926B-B3D1580E9851%7D&page-id=%7B54BA5546-8EDC-4553-A0FC-6AA90B4ADD09%7D&object-id=%7B58A09568-CFF8-43F9-84E8-1F2362390883%7D&E

jlucas9, 2022-11-15:
Closing for now - will be revisiting in new issues.

closed 2022-11-15 by jlucas9