nasa/nos3 #83: Run NOS3 Example with COSMOS and 42 on different VM

kabarr, 2022-09-22:
I've created 2 VMs and successfully changed settings to run COSMOS and 42 on one machine and the Sims on another.  I've run into a snag that may not be possible to fix.  The truth42Sim data is all 0's in the sim, and the 42 satellite location is "static".  I suspect this is due to me not understanding all the connections between 42 and the sim.  Is it possible to run 42 on a different machine?

jlucas9, 2022-09-23:
Hi @kabarr,
It should certainly be possible to run them in different VMs. I'm working now on dockerizing the simulators in issue #78, but it's a slow process. 42 should be directly feeding data to the simulators via the [sim_common/src/sim_data_42_socket_provider.cpp](https://github.com/nasa-itc/sim_common/blob/e3eb4d7fb3347e838a91055722548ac4c86cc461/src/sim_data_42socket_provider.cpp) interface. The simulators will timeout if they cannot connect to 42 if I remember correctly - maybe that's what's occurring?

kabarr, 2022-09-26:
It appears that the 42 truth data is all 0's.  I may not be running it on the correct VM.

VM1: With Flight SW and Sims:
***@***.***

VM2 with 42 Running (sat position does not update:
***@***.***

jlucas9, 2022-10-07:
Quick update - I have been able to duplicate this issue on my side and am working with some other experts to debug!

jlucas9, 2022-10-11:
^ From the `nos3#83` branch 7a66dab:
* `vagrant up` to deploy the VM
* `vagrant halt` to shut it down
* Inside the Oracle VM VirtualBox Manager
  * Tools > Network > Add
    * Network Name: NOS3
    * Network CIDR: 192.168.100.0/24
  * Edit the `nos3_1.5.2` settings
    * Network > Adapter 1 > NAT Network > NOS3
  * Manually cloned the `nos3_1.5.2` VM in the Oracle VM VirtualBox Manager
  * Updated cloned VM settings:
    * Network > Adapter 1 > Advanced > Adapter Type > Intel PRO/1000 MT Server
    * Note that this is different than the default provided to ensure it gets recognized as a different on the network

To summarize I've got:
* NOS3 NAT setup in VirtualBox
* VM0 - FSW and Simulators
* VM1 - 42 and COSMOS (not setup)

I am getting valid 42 data in the truth simulator. If we want to get COSMOS running too in this branch let me know and I can fix it up.

kabarr, 2022-10-17:
Thanks John!  I'll give this a try - I was out last week (forgot to set my "away notice...).

Best Regards,
Kathy

kabarr, 2022-10-19:
I was able to review the changes in the #83 branch and update the 2 VMs I had already created.  I'm now seeing Truth data on the machine with FSW/Sims and the 42 is updating on the VM with COSMOS.  Thanks!

jlucas9, 2022-10-20:
Final step for this issue is to (from the new `dev` branch 59e8ec6) recreate the above and get COSMOS command and telemetry working. @msuder

kabarr, 2022-10-20:
As an FYI, I have COSMOS command and telemetry working on the "ground station" VM already, so the help with 42 was the only thing I needed.  I'm interested in seeing your solution since mine may not be the best approach.

msuder, 2022-10-21:
nos3 repo, branch nos3#83-DualVM (with submodule cosmos branch nos3#83-DualVM) has been configured so that flight software (FSW) and simulators run on one VM with IP address 192.168.100.4 and 42 and COSMOS run on another VM with IP address 192.168.100.5.

The following files/commands have been modified so that FSW/sims run on 192.168.100.4 and 42/COSMOS run on 192.168.100.5:
- sims/cfg/nos_engine_server_config.json
- - configure the nos3 server URI to use 192.168.100.4
- sims/cfg/nos3-simulator.xml
- - configure the nos-connection-string to be 192.168.100.4:12001
- - configure the data-provider hostname to be 192.168.100.5 for each simulator (to connect to 42)
- sims/cfg/InOut/Inp_IPC.txt
- - configure the server hostname to be 192.168.100.5 (for each 42 port that is used for a simulator)
- sims/cfg/InOut/Inp_NOS3.txt
- - configure the NOS3 time connection string to use 192.168.100.4 (42 time connection)
- config/tools/cmd_tlm_server/nos3_cmd_tlm_server.txt
- - configure the MISSION_INT interface to connect to 192.168.100.4 (COSMOS to FSW/CI_LAB/TO_LAB connection)
- COSMOS command:  CFS TO_ENABLE_OUTPUT with DEST_IP '192.168.100.5', DEST_PORT 5011
- - configure FSW/TO_LAB to send telemetry to COSMOS at IP 192.168.100.5

msuder, 2022-10-21:
Useful for assigning static IPs to VMs:  https://ernieleseberg.com/virtualbox-with-static-ip-on-ubuntu-server-guest/

msuder, 2022-10-21:
**QUAD VMs:** nos3 repo, branch nos3#83QuadVM (with submodule cosmos branch nos3#83-DualVM) has been configured so that flight software (FSW) and NOS Engine Server run in one VM with IP address 192.168.100.4; sims run in a VM with IP address 192.168.100.5; 42 runs in a VM with IP address 192.168.100.6; and COSMOS runs in a VM with IP address 192.168.100.7.

The following files/commands have been modified for this:
- sims/cfg/nos_engine_server_config.json
-- configure the nos3 server URI to use 192.168.100.4
- sims/cfg/nos3-simulator.xml
-- configure the nos-connection-string to be 192.168.100.4:12001
-- configure the data-provider hostname to be 192.168.100.6 for each simulator (to connect to 42)
-- configure the COSMOS IP for the truth simulator to be 192.168.100.7
- sims/cfg/Inout/Inp_IPC.txt
-- configure the server hosname to be 192.168.100.6 (for each 42 port that is used for a simulator)
- sims/cfg/InOut/Inp_NOS3.txt
-- configure the NOS3 time connection string to use 192.168.100.4:12001 (42 time connection)
- config/tools/cmd_tlm_server/nos3_cmd_tlm_server.txt
- COSMOS command:  CFS TO_ENABLE_OUTPUT with DEST_IP '192.168.100.7', DEST_PORT 5011
-- configure FSW/TO_LAB to send telemetry to COSMOS at IP 192.168.100.7

jlucas9, 2022-11-29:
Closing for now as new issues will be made to reference this one and further expand on the capability.

closed 2022-11-29 by jlucas9

Simon9050, 2024-02-27:
Hi,

First off, big thanks for your work on NOS3 simulator! It’s a game-changer for the space community.

I tried running NOS3 in dual VM mode, following the instructions, but it didn’t quite cooperate... Any tips or a specific branch I should use for this setup?

Cheers, 

Simon