nasa/nos3 #84: Option for fprime

jlucas9, 2022-09-23:
https://github.com/nasa/fprime

Should be able to support applications in fprime as well as cFS and have the option to run either.

EbenezerA99, 2024-02-27:
Hi ! Have there been any updates on possible integration with Fprime?

jlucas9, 2024-02-27:
Hi @EbenezerA99,
We're working on getting this assigned to one of the NOS3 developers here in the coming week and starting the work! We'll keep you posted as we go along here and in the associated branch once it's created.

aerospaceadversarylab, 2024-02-28:
@jlucas9 Is there anyway our lab at Cornell University can contribute to this? Would be excited discuss further (email: gfalco@cornell.edu).

zlynch2, 2024-04-11:
Hi @aerospaceadversarylab,
Below is a list of steps on how to checkout out our current progress on the fprime integration. The goal of theses steps is to see fprime-gds send our noop command come through on our samplesim terminal on the nos3 side. We do this by hijacking the fprime helloworld example and modifying to include our nos3 stuff. Let me know if you have any questions thanks!

It is assumed that you have a nos3 VM, inside clone the reference repos somewhere under /home/jstar to try to avoid permissions issues. For example, make a directory called dev and clone the 2 repos inside it.

1. Pull Docker deployment branch **nos3#84-fprime-checkout** from https://github.com/nasa-itc/deployment
2. Build Docker container by running **docker build -t ivvitc/nos3-64:fprime .** 
	a. Make sure to be in the **deployment/** directory
3. Pull Nos3 repo branch **nos3#84-fprime-checkout** from https://github.com/nasa/nos3/tree/nos3%2384-fprime-checkout 
4. Run **git submodule update --init --recursive**
5. Run **make prep** (ignore fprime not found error we have this locally)
6. Run **make**
7. Cd nos3/components/sample/fprime/fprime-nos3/
8. Run **Mkdir fprime-venv/**
9. Run **python3 -m venv fprime-venv/**
10. Cd nos3/
11. Run **make checkout**
12. Open a new terminal
13. Ensure you are in nos3/ and run **make debug**
	a. This will enter the docker container
14. In the docker container cd into components/sample/fprime/fprime-nos3
15. Run **.  fprime-venv/bin/activate**
16. Run **pip install -r fprime/requirements.txt**
17. Run **fprime-util generate**
18. Run **fprime-util build**
	a. Note you will only have to generate and build once (as of now, since we have a script to build later)
19. cd nos3/
20. Run **./scripts/fprime_launch.sh** 
21. In your other terminal (jstar@itc:~/dev/nos3) outside the docker container run **docker container inspect nos3_debug**
	a. Copy the IP address (Ex. 172.25.0.6)
22. Open firefox and type http://[your_IP_Address]:5000/ (ex. http://172.25.0.6:5000/)
	a. At this point you should see the fprime-gds running and connected
23. Under the commanding tab in the fprime-gds send a say_hello command
	a. You should see the noop come down after a couple seconds through your samplsim terminal in Nos3

zlynch2, 2024-04-16:
Hi @aerospaceadversarylab ,

We made some slight edits to the build process which may have changed the steps above. Use the below. The goal is still the same however we completed the two way communication between Nos3 and Fprime. We also automated the process a little for you with our scripts.

It is assumed that you have a nos3 VM, inside clone the reference repos somewhere under /home/jstar to try to avoid permissions issues. For example, make a directory called dev and clone the 2 repos inside it.

1. Pull Docker deployment branch **nos3#84-fprime-checkout** from https://github.com/nasa-itc/deployment
2. Build Docker container by running **docker build -t ivvitc/nos3-64:fprime .** 
	a. Make sure to be in the **deployment/** directory
3. Pull Nos3 repo branch **nos3#84-fprime-checkout** from https://github.com/nasa/nos3/tree/nos3%2384-fprime-checkout 
4. Run **git submodule update --init --recursive**
5. Run **make prep** (ignore fprime not found error we have this locally)
6. Run **make**
7. Cd nos3/components/sample/fprime/fprime-nos3/
8. Run **Mkdir fprime-venv/**
9. Run **python3 -m venv fprime-venv/**
10. Cd nos3/
11. Run make checkout
12. Open a new terminal
	a. A Fprime terminal will open automatically and start building, you can use this as well just exit out of the current process (ctrl c or exit)
13. Ensure you are in nos3/ and run **make debug**
	a. This will enter the docker container
14. In the docker container cd into components/sample/fprime/fprime-nos3
15. Run **.  fprime-venv/bin/activate**
16. Run **pip install -r fprime/requirements.txt**
17. Run **fprime-util generate**
18. Run **fprime-util build**
	a. Note you will only have to generate and build once (as of now, since we have a script to build later)
19. In your other terminal (jstar@itc:~/dev/nos3) outside the docker container run **make stop**
	a. This will close all terminals and stop their docker containers
20. Finally run **make checkout**
	a. This should automatically launch a fprimg-gds in firefox for you
21. Under the commanding tab in the fprime-gds send a **sampleSim.NOOP** command
	a. You should see the noop come down after a couple seconds through your samplsim terminal in Nos3
	b. You can also send sampleSim.Request_Housekeeping and see telemetry come back from Nos3 under the channels tab. After sending the command you will see sampleSim.DeviceCounter increment under channels tab in fprime-gds.

aerospaceadversarylab, 2024-04-17:
@zlynch2 Thanks a lot for the steps, we will try this out and report our findings to you during our next meeting.

emmahanson7, 2024-08-21:
Hi! We're looking into using this for our lab at Georgia Tech, what is the current status of the development? Is it relatively stable at this point?

zlynch2, 2024-08-22:
Hi @emmahanson7 !

We are currently working towards a stable version integrated in nos3 for our next release and/or point release. Keep an eye on this issue as you should see progress in the near feature. We still have a couple items on our personal checkout to complete. We will most likely close this issue when we merge into dev.

jlucas9, 2024-10-17:
Merged into `dev`!

closed 2024-10-17 by jlucas9

zlynch2, 2024-10-17:
This issue is considered completed, new issues concerning fprime will be created as we continue to checkout NOS3 generic components into fprime fsw. Sample is running in nos3 with fprime in minimal spacecraft mode and is integrated into the dev branch of nos3.