nasa/nos3 #110: HWLIB Review

jlucas9, 2023-01-04:
Need to review and update the HWLIB to address the following:
- [  ] The CW_addr logic in the libcan.h and sim can be removed

jlucas9, 2023-01-20:
## Libtrq

From internal chat - 

Also, I'm rethinking my comment about hwlib a little... the functions declared in the header seem reasonable except for 2 things:  1.  why does trq_set_period not take a period input parameter? 2.  I might give some thought to whether there should just be either trq_command or the triple trq_set_time_high, trq_set_period, trq_set_direction ... I think they are almost redundant

Ok... here is what i don't like about the trq_info_t struct - any code outside of hwlib/libtrq.c can alter it... and in fact it is assumed that its 1st 2 members are altered prior to calling trq_init... which leads me to not liking trq_init... you should probably just specify what torquer to initialize and then this function should return you a trq_info_t that you use for the other calls (like open does for read/write/close)

So there is one more problem there... how do you get timer_period_ns set? right now it is done prior to calling trq_init... but again, nothing prevents code outside of hwlib/libtrq.c from altering it later. And what are the valid values for this? Is it fixed in the datasheet for the torquer or a range in the datasheet? In the first case it could then be set in trq_init... in the second case nothing is doing that kind of checking which ought to be done

I think there is a reason open gives you an "int fd" and not a pointer into all the internals of what is going on with a file/device/etc. Then you have to use the other functions (read/write/close/ioctl/etc.) that only take an "int fd"... you can't bypass them and alter the structure directly

msuder, 2023-02-14:
More hwlib thoughts:
1. Assertion:  hwlib should be the only part of flight software that "knows about" (interfaces with) NOS Engine
2. Possible? Assertion:  hwlib should not be used by component simulators, it should only be used by core flight software and component flight software and is used to abstract away hardware bus (and other, e.g. GPIO?) interfaces
3. Comment:  hwlib/fsw/public_inc/lib*.h files are and should have very well thought out/stable interfaces:  These provide the mechanism for easy switching between NOS Engine and real CPU/SBC/Board hardware (potentially of multiple flavors... cFS supports multiple targets)
4. Assertion:  hwlib should not be hardcoded to have a fixed number of each type of bus (if possible)... perhaps defer this part of this issue to another issue for later implementation
5. Assertion:  All hwlib communication to NOS3 simulators should be via NOS Engine Bus (add more "bus" or "pseudo-bus" types if necessary, such as GPIO)

msuder, 2023-02-20:
Another thought about the torquer section of the hardware library.  The existing code kind of entangles the functionality of a piece of hardware (the specific magnetorquer) with the communication protocol for it.  I think we should abstract the functionality into:
1.  A pulse width modulation (PWM) protocol, with parameters to set the (GPIO) pin, the PWM period, and the PWM time high
2. A GPIO protocol, with parameters to set the GPIO pin and its state (high/low).

msuder, 2023-02-20:
Enabled/Disabled/Open/Closed etc. need thought out for relevance and consistency to each hwlib protocol.

msuder, 2023-02-20:
Change GPIO to use NOS Engine (probably base bus type).

jlucas9, 2023-03-20:
Need to add CRC generation and check to CAN as a part of HWLIB since it is the standard - https://en.wikipedia.org/wiki/CAN_bus

msuder, 2023-03-20:
So SPI is a weird option in hwlib due to needing to specify a bus and a chip select. so spi_XY is where X is bus and Y is CS. then you also have to define the bus/cs defines to match that.

We should alter the hwlib sometime to split out the busname and the CS for SPI

msuder, 2023-03-22:
Redo libcan to not be specific to CubeWheel.

mgrubb-gsfc, 2023-06-20:
This is ready to merge when the Arducam and Novatel tickets are closed out. Those two components will need minor updates to conform to the new API. Currently the branch has those removed from the build and the _defs.

For other comments above, CAN lib was updated to remove specific for the CW. SPI was updated to use bus and cs. Open/Close was fixed and made consistent for all busses, can now open and close and NE will reconnect. 

Not adding GPIO or PWM to use engine since those are electrical interfaces that are not passing bytes. GPIO uses the devices like linux would, and PWM in this case is only used for the torquers. Leaving lib torque as its own lib since it is a specific case to drive torquers and has heritage.

closed 2023-07-10 by jlucas9