nasa/nos3 #111: Cannot install on mac M1

himilcon, 2023-01-05:
Hello,
I have tried to install NOS3 in my mac, following the instructions
Virtualbox (v.7.0) ubuntu machine quits in "aborted" state after "vagrant up":

himilcon [nos3]  % vagrant up
Bringing machine 'default' up with 'virtualbox' provider...
==> default: Checking if box 'nos3/ubuntu' version '0.0.0' is up to date...
==> default: Clearing any previously set forwarded ports...
==> default: Clearing any previously set network interfaces...
==> default: Preparing network interfaces based on configuration...
    default: Adapter 1: nat
==> default: Forwarding ports...
    default: 22 (guest) => 2222 (host) (adapter 1)
==> default: Booting VM...
==> default: Waiting for machine to boot. This may take a few minutes...
    default: SSH address: 127.0.0.1:2222
    default: SSH username: vagrant
    default: SSH auth method: private key
    default: Warning: Connection reset. Retrying...
The guest machine entered an invalid state while waiting for it
to boot. Valid states are 'starting, running'. The machine is in the
'aborted' state. Please verify everything is configured
properly and try again.

Any clue on what is going on here?
Thanks,
Himilcon

jlucas9, 2023-01-05:
Hi @himilcon,
I'd suggest a couple things to try first:
* Confirm you're on the the `main` or `dev` branch of NOS3
  - git status
* Clean up install and try again to confirm it's a repeatable error
  - vagrant destroy
  - vagrant box update
  - vagrant up

Also, have you run virtual machines previously on this machine?

himilcon, 2023-01-05:
Hello jlucas9,
I'm on the main branch and have tried the a new install with no success.
I think the problem could be related to the fact that the nos3/ubuntu machine might not be compatible with the ARM64 architecture of the M1 mac.
Or could be VitualBox being in a Beta version for Mac M1...
Everything went well in my old Intel chip Macbook

jlucas9, 2023-01-06:
Hi @himilcon,
I was able to track down a new M1 and run a quick test. Unfortunately I can confirm the same issue with Virtualbox 7.0.4 (beta 4) and Vagrant 2.3.4. In the vagrant error it strongly points the finger at VirtualBox, so I tried launching the already downloaded box from VBManager and still have it crashing even after changing all settings available. Seems we may have to wait on an updated beta or final release version of VirtualBox for the M1.

jlucas9, 2023-01-17:
Closing this issue for now, can re-open if issue persists after VirtualBox has a release for M1 hardware

closed 2023-01-17 by jlucas9