nasa/nos3 #112: User Improvments

jlucas9, 2023-01-06:
Request to have the following installed by default:
* Ubuntu
- [ ] Desktop scripts
- [ ] Task bar
* Rocky Linux
- [ ] Desktop scripts
- [ ] Task bar

jlucas9, 2023-02-02:
@msuder should we tackle this also before the next base box release?

msuder, 2023-02-02:
Yes, I think so... I'm not a fan of the current baseboxes not having these things.

msuder, 2023-02-07:
Worked in this issue:
https://github.com/nasa/nos3/issues/116
and this commit:
https://github.com/nasa-itc/deployment/commit/0ebd8d4668c7e320e405717e4278af16eabf4779

msuder, 2023-02-08:
This was closed by merge, commits:
https://github.com/nasa-itc/deployment/commit/2a0d258c2f8462cc8c6ab21b582ef57d3e55b7de
https://github.com/nasa/nos3/commit/a285150cc2d282c39ee6a2b0cf52d3ea3e656a21

closed 2023-02-08 by msuder