nasa/nos3 #117: Sim Terminal Write

jlucas9, 2023-01-24:
In `simulator_terminal.cpp` the “WRITE” command does not write the end of string terminator.
This may cause a sim that is receiving the command to see “extra” characters.

Suggested Fix: 
* Add 1 to wlen in the _bus_connection->write call – this has worked consistently for me.

[image]

jlucas9, 2023-01-24:
Possible same issue with the wlen (vs wlen+1) using the TRANSACT vs WRITE command in the sim terminal. 
There may be others, but I’ve only used the WRITE and TRANSACT.

jlucas9, 2023-02-07:
While the +1 length in sim terminal resolves this issue for this case, it will cause issues for simulators that communicate by sending ASCII characters over a specific protocol (like the GPS). Another solution is to have each simulator:
* Check commands against a substring
  - `else if (command.compare(0,6,"ENABLE") == 0)`
* Add a readline to responses
  - `response = "SampleHardwareModel::command_callback:  Enabled\n";`

msuder, 2023-02-08:
Perhaps the sim terminal ascii mode should support \ escape characters so the user could enter a \0 to null terminate the string... or perhaps the sim terminal should be more "protocol" aware and know to do something "different" (null terminate ascii strings?) for "command" bus types... or/both... food for thought.  These ideas might belong in separate issues.