nasa/nos3 #126: NOS Time Driver - Send ticks to multiple NOS engine servers for distribution

msuder, 2023-02-08:
Motivation:  Run multiple spacecraft by running multiple instances of nos3, each with a single spacecraft... and with each nos3 instance having its own NOS Engine Server.  Would like to have a single source of truth time for all spacecraft, so time ticks need to be sent/synchronized to all NOS Engine Servers.

Possible designs:
1. Modify NOS Time Driver to be able to connect to multiple NOS Engine Servers and send a tick to all of them at once.
2. Create a new executable which connects to current style NOS Time Driver connected to each NOS Engine Server... new executable sends ticks to current style NOS Time Driver which forwards them out through NOS Engine Server (so instead of the current NOS Time Driver deciding itself when to send a tick, it would not send a tick until the new executable commanded it to).  (Need to think about acknowledgement if this design is used... I believe current NOS Engine design provides acknowledgement from time slaves back to the (single) time master.)

msuder, 2023-02-10:
Design 1 was chosen and was fairly simple to implement within the context of the current time driver.

Clarification of original comment:  Note that the time ticks really go to a bus on each NOS Engine Server, not to the NOS Engine Server itself.

A few notes on NOS Engine design for time master(s) / client(s):
1. A bus can have at most one time master, but many time clients
2. Each bus can have its own time master
3. A bus with time clients but no master will not get ticks even if another bus on the same NOS Engine Server is getting ticks from a time master.
4. If you want multiple busses to get the ticks from a single time master, you can create a NOS Engine BusGroup and use the set_time method of the group (although this is not what was done for this issue, because we needed to send ticks to busses that were associated with different NOS Engine Servers).

jlucas9, 2023-03-08:
Confirmed fully functional and merged into `dev`!

closed 2023-03-08 by jlucas9