nasa/nos3 #131: Does Manipulating 42 GUI Alter NOS3 Time and/or 42 Data Output (Rate) and/or Sim Behavior?

msuder, 2023-02-10:
John:    Not really a FSS issue, but it seems that if I am playing with the camera in 42 weird things happen with the reported data. I still get telemetry from cFS, but the reported data from the sim basically freezes - I added HK and data to the scheduler for this test.
Mark:  That would be good to run down a little... I am guessing that the issue is with 42 stopping/slowing its execution loop calculations when the GUI is being changed... but I'm not sure.

Steps:
1. Research what is going on.  
2. Determine if the behavior is "correct" or "incorrect".  
3. If "correct", consider documenting the behavior and noting that it is correct.
4. If "incorrect", determine if anything can be done or if it is a 42 (or other) limitation.  
5. If "incorrect" and something can be done, do it.
6. If "incorrect" and nothing can be done, document limitation.

Isgaroth-the-Green, 2023-04-06:
Looking at the issue, it would appear that the spacecraft stops moving entirely/freezes when the camera is being adjusted in 42. By starting up the torquers (to get the spacecraft spinning) and then looking at the values returned by the FSS, it becomes apparent that when the camera is moving, the spacecraft stops entirely - the torquers do not cause it to rotate, and the FSS therefore sends the same packets. When I rotate the camera and the FSS is pointed towards the sun, it continuously gives the same packets as it did before I started the camera rotation; when it is pointed away from the sun, it does the same.

jlucas9, 2024-06-14:
On the latest `dev` branch it seems like both the GENERIC_ADCS and SIM_42_TRUTH (i.e. 42) both still report telemetry but nothing actually moves forward in time.  Closing for now, but should continue to develop tests for this as we get unit and system level tests standardized in NOS3.

[image]

closed 2024-06-14 by jlucas9