nasa/nos3 #135: Transition nos3 Repository to be an Integration Only Repository Full of Submodules (and a few Integration Files)

msuder, 2023-02-14:
Integration includes submodules for:
1. fsw (cFE/cFS)
1. gsw (currently AIT/COSMOS/OIPP... maybe each should be its own repository)
1. sims (common functionality + time driver + sim terminal + truth_42_sim (maybe this should be its own repository, maybe a component?))
1. components/ - A directory containing 1 or more submodules conforming to the NOS3 component repository structure (README/fsw/gsw/sim)

What else should be in the nos3 repo?
1. Top level build time configuration
    1. Prefer CMake/CMakeLists.txt/etc. to the current Makefile (currently everything is CMake compatible except maybe some of fsw)
    1. Figure out how to specify
        1. The base fsw/gsw/sims to build + 
        1.  How to get to the submodule build configuration:  All the components/*/fsw, components/*/sim, components/*/gsw to build
1. Top level run time configuration - Figure out how to specify
    1. The base fsw/nos3_defs/*.scr?, sims/cfg/nos3-simulator.xml<common/> files that provide an integrated top level system configuration
    1. How to get to the submodule configuration:  All the components/*/sim/cfg, components/*/gsw files that provide the component configuration that need included into things like nos3-simulator.xml<simulators/> and 42 InOut/* and gsw/cosmos/config/system/nos3_system.txt and gsw/cosmos/config/tools/cmd_tlm_server/cmd_tlm_server.txt  files that provide component configuration within the top level system configuration
1.  Top level run/stop (and build/clean?) scripts
    1.  What do we do with things in gsw/scripts?
1.  What do we do with fsw/nos3_defs?  Could fsw be a stable "distribution" of cFS for NOS3 with that removed?

jlucas9, 2023-04-13:
[image]

msuder, 2023-04-27:
[SSPOMissionRepoBuildDeploy.docx](https://github.com/nasa/nos3/files/11344280/SSPOMissionRepoBuildDeploy.docx)

jlucas9, 2024-01-18:
Merged to `dev`

PR #240 (2024-01-18): [nasa/nos3#135] Configuration
Updated the file structure to have configuration files in a top level `cfg` directory.
These XML configuration files are used by the `configure.py` script to update the default InOut (42), nos3_defs (FSW), and sims files.

Closes #135
closed 2024-01-18 by jlucas9