nasa/nos3 #136: Add Deconflicted List of Performance IDs to Wiki Component Development Page

msuder, 2023-02-15:
https://github.com/nasa/nos3/wiki/NOS3-Component-Development

This page provides a nice, deconflicted list of cFS message ids allocated to the various nos3 provided components.

It would be nice for it to provide the same for cFS performance ids.

jlucas9, 2023-02-21:
Does cFS 7.0 use perf ids still? The tools are nice to use if they still exist

kevincbruce, 2024-02-29:
I'd be happy to look into it. Could use a refresher on perf IDs. Also might change the name of that page to _'Component Directory'_ or something like that, since it is more a directory or list of components and IDs than it is a development guide (the Component Development Flow page _is_ a dev guide meanwhile)

kevincbruce, 2024-03-01:
Okay, went through manually today. Added perf_ids to the components, and made a section for CFS apps and added Message IDs and Perf Ids there, as applicable. Let me know what you think. Also, are there Perf IDs for Cryptolib? I couldn't find them, nor the message IDs that were already referenced. I'll also check through to make sure the MIDs that were listed for components are accurate.

kevincbruce, 2024-03-04:
Due to offset 30+x and 40+x, CF still collides, but this should be resolved for all unique, set IDs, and the wiki should be up to date to match the current reality once the branch is merged. Issue is worth reviewing with the meetings around and after the next update, when we look at updating CF.

kevincbruce, 2024-03-04:
And confirmed that it builds and seems to work correctly with the updated list.

jlucas9, 2024-03-06:
Merged into `dev`!

closed 2024-03-06 by jlucas9