nasa/nos3 #138: Does COSMOS Need Separate RADIO vs. DEBUG Targets?

msuder, 2023-02-16:
@[msuder](https://github.com/msuder) msuder [2 days ago](https://github.com/nasa-itc/gsw-cosmos/pull/6/files/f0f0e12f3f7ab589ca91e70c579a589b6772d088#r1105924781)
For now this is ok. Comment/question for the future: Why are there "RADIO" and non-"RADIO" versions? Does this cause any duplication that is unnecessary (DRY=Don't Repeat Yourself)? How do we/can we manage the multiple COSMOS/gsw "interfaces" (lab vs. radio vs. ??) to flight software (ci_lab/to_lab, ci/to, etc.)?

Member
Author
@[jlucas9](https://github.com/jlucas9) jlucas9 [1 hour ago](https://github.com/nasa-itc/gsw-cosmos/pull/6/files/f0f0e12f3f7ab589ca91e70c579a589b6772d088#r1108529102)
The issue here is that the same target cannot be assigned to multiple interfaces so I've been using the initial definition for "debug" and then creating another for "_radio". I'm sure we can devise a better way for this as we move forward.

jlucas9, 2023-07-10:
Could be resolved as part of issue #175

ryanmelt, 2023-08-06:
COSMOS 5 supports connecting the same target to multiple interfaces for telemetry.  For commands you can as well, but will typically just want one interface at a time