nasa/nos3 #141: Further Generalize/Abstract the sim_common SimData42SocketProvider/Sim42DataPoint

msuder, 2023-02-17:
[Tuesday 8:48 AM] Lucas, John P. (IVV-1810)
The string we parse from 42 itself and information about it - I think the generic parsing functions could be moved up to sim_common (eventually) so that new developers don't really need data provider knowledge and we could expand to other data providers without adding in specific functionality to each component

...

... on the other hand, it would not be very hard to implement a "42 blackboard"... hmmm

msuder, 2023-02-17:
A few current architecture/design notes:
1.  Interface from SimData42SocketProvider to 42 is generic (virtually nothing needs done by any specific sims derived class of this class) - decoupled well
2. Complication is that the block of text lines to/from 42 can change (either by filtering in the 42 Inp_IPC.txt file or by 42 changes over time) - coupling pain point
3. Currently, this has led to 2 things:  
1.  Locked down version of 42 used in NOS3 VM (currently https://github.com/ericstoneking/42/commit/f20d8d517b352b868d2f45ee3bffdb7deeedb218) - address coupling pain point
2. Each sim defining the parsing (in Sim42DataPoint derived class) of the data the hardware model needs - coupling weakness in NOS3

Design:
1. Move parsing of 42 "message" (block of ASCII text delimited by time at the start and the keyword "END" at the end iirc) from each individual sims data point back into the Sim42DataPoint base class
2. How:  Parse each line to split on the delimiter "="... use a map to store the first part as the key and the second part(s) as the value
3. Hardware Models can lookup data they need by "key"
4. N.B.  Second part(s) can be one or more value depending on the key... easily split by delimiter " " into a vector... harder is to "infer" the data type of each vector value... needs more thought
5. Even the key is splittable to extract information (e.g. "SC[0]." prefix tells you which spacecraft the data point is for... and a 42 "message" can have multiple spacecraft (and orbits, and worlds, ...) in it) ... do we need this information (yet?)... might be easy enough to generalize that it is worth doing now anyway (42 is very consistent, even from version to version, about at least some parts of the key)

Value of the new design:  Better decoupling (and abstraction) of hardware model/data provider architecture provided by sim_common.

jlucas9, 2023-08-01:
Merged into `dev`

PR #208 (2023-08-01): 141 further generalizeabstract the sim common simdata42socketprovidersim42datapoint
Depends on PRs:

- [ ] https://github.com/nasa-itc/generic_css/pull/7
- [ ] https://github.com/nasa-itc/generic_fss/pull/5
- [ ] https://github.com/nasa-itc/generic_imu/pull/6
- [ ] https://github.com/nasa-itc/generic_magnetometer/pull/6
- [ ] https://github.com/nasa-itc/generic_reaction_wheel/pull/7
- [ ] https://github.com/nasa-itc/novatel_oem615/pull/8
- [ ] https://github.com/nasa-itc/sample/pull/11
- [ ] https://github.com/nasa-itc/sim_common/pull/7

Closes #141.
closed 2023-08-01 by jlucas9