nasa/nos3 #142: NOS Engine Server Error on nos3 Launch

msuder, 2023-02-17:
NOS Engine Standalone Server

Reading Configuration File: /home/nos3/Desktop/git/github.com/nasa/nos31/sims/build/bin/nos_engine_server_config.json

Loading Plugins: 
  uart

Creating Transports:
  fsw: tcp://127.0.0.1:12000
  nos3: tcp://127.0.0.1:12001

Query Menu:
1. Buses
2. Data Nodes
3. Time Sender
4. Time Clients
5. Exit
**Command: 2023-02-17 13:40:10.433591 [ ERROR ] NosEngine.Uart - close uart port failed: node (fsw) not connected to port (2)
2023-02-17 13:40:10.444930 [ ERROR ] NosEngine.Uart - close uart port failed: node (fsw) not connected to port (1)**

Why does this error appear?  Is it a NOS3 bug?

Steps to reproduce:
1.  Use commit https://github.com/nasa/nos3/commit/ecd65c0e0cdadb33d6939c50bfa8325b3a5d5970 (dev branch points there at the moment)
2. Run the typical make, make launch
3. This error crops up during the initial startup of all the sims, fsw, 42, COSMOS

jlucas9, 2024-03-19:
This actually got way worse with the latest nos3-64 bit docker container using the release versions of NOS3. Need to get these errors reclassified.

jlucas9, 2024-06-14:
Working in #231, can confirm warnings have been suppressed:

[image]