nasa/nos3 #143: NOS Engine Server / Sim Terminal Error on switch to SPI Bus

msuder, 2023-02-17:
NOS Engine Server window:
```
2023-02-17 13:56:51.746555 [ ERROR ] NosEngine.Common - receive operator failed to receive message, transport not connected, tcp://localhost:45792 in class)::operator()(const boost::system::error_code &, size_t) const at transport/src/Transport/TCPConnection.cpp:386
```

NOS Terminal window (sim terminal):
```
SimTerminal:<terminal@command>:Node:<sample-sim-command-node>:Mode:<IN=ASCII:OUT=ASCII> $ set simnode 1
SimTerminal:<terminal@command>:Node:<1>:Mode:<IN=ASCII:OUT=ASCII> $ set simbus spi_1
2023-02-17 13:56:51.745386 [ DEBUG ] - BaseConnection: deleting old base connection Handle
2023-02-17 13:56:51.745699 [ ERROR ] NosEngine.Common - receive operator failed to receive message, connection was shutdown in NosEngine::Transport::Connection::process_stop(std::unique_lock<std::mutex> &) at transport/src/Connection.cpp:377
2023-02-17 13:56:51.745880 [ ERROR ] NosEngine.Common - receive operator caught exception while receiving message, will stop receiving, message connection was shutdown, before async receive message started
SimTerminal:<terminal@spi_1>:Node:<1>:Mode:<IN=ASCII:OUT=ASCII> $ 
```

Steps to reproduce:
1.  Use commit https://github.com/nasa/nos3/commit/ecd65c0e0cdadb33d6939c50bfa8325b3a5d5970
2. Run typical make, make launch
3. In sim terminal (NOS Terminal window) enter these commands:
```
SimTerminal:<terminal@command>:Node:<sample-sim-command-node>:Mode:<IN=ASCII:OUT=ASCII> $ set simnode 1
SimTerminal:<terminal@command>:Node:<1>:Mode:<IN=ASCII:OUT=ASCII> $ set simbus spi_1
```
4.  Errors specified above immediately appears in both windows

jlucas9, 2024-06-14:
Note that still occurring in #231 after other updates to common/engine:

[image]

[image]

jlucas9, 2025-03-26:
Backlog as switch to QEMU make make NOS Engine obsolete.