nasa/nos3 #145: ADCS Sensors and Actuator Phasing Test and Verification

msuder, 2023-02-20:
During development and testing of  Generic ADCS (https://github.com/nasa/nos3/issues/101) and during integrated spacecraft testing (simulation only for the generic case), the polarity of various ADCS sensors and actuators must be double-checked between actual hardware and the hardware simulators that represent actual hardware.

This issue will document that testing and drive hardware simulation changes needed to resolve any discrepancies.  

In this case of generic hardware components and generic ADCS, this verification may need to be done via code inspection since real hardware may not exist.

msuder, 2023-02-20:
Related to #101 .

jlucas9, 2023-04-27:
It seems that when the TRQ is set to 100% duty cycle (always on) you don't see this in the MAG data.
Does 42 not have actuators directly influence sensors?

kevincbruce, 2025-03-26:
@jlucas9 is this a priority?

jlucas9, 2025-03-26:
@kevincbruce yeah, I think we cover this as part of the testing for LPT / CPT