nasa/nos3 #150: SimTerm - UDP Interface

jlucas9, 2023-03-08:
Add a UDP interface to Sim Term to enable other means of commanding instead of just CLI.

msuder, 2023-03-17:
Design thoughts:

1. Should the terminal interface be a data provider?  
2. Should the UDP interface be a data provider?  
3. Should there be a "file" interface (to feed a file of commands to sim terminal) and should it be a data provider?

Current thought:  
- No, don't make them data providers; this would unnecessarily complicate the design and would really stretch the data provider concept.  
  - The current data provider concept for all the other sims is that the hardware model is the one that needs data and then it queries the data provider for the data.  
  - In the sim terminal case the terminal/UDP/file is what sends data first and then the hardware model would have to handle that data, do the command(s), and reply back to the terminal/UDP/file; so a different sort of concept.

Further thought:  
- Develop the interfaces not as data providers for now, but leave open the possibility of changing them to data providers at a later time if that makes sense (the data providers framework enables data providers to be plug-ins and to more easily dynamically instantiate one or more of them).

Final thought:  Still not sure about adding a "file" interface... I can see a lot of potential use for it, but there are probably some design issues to work through (e.g. if/how to control the "timing" of when commands are sent).

msuder, 2023-03-22:
Not going to add a file interface right now.  This might be able to be done "on the cheap" as follows (assuming a sim terminal of UDP type has been started and is listening on port 5556):

input.txt:
set prompt short
set hex
transact 56 asdfasdf

bash shell:
$ while read -r line; do echo -n $line > /dev/udp/localhost/5556; done < input.txt

PR #152 (2023-04-03): 150 simterm udp interface
Will need the following closed first:
- [x] https://github.com/nasa-itc/sim_terminal/pull/1

Closes #150
closed 2023-04-04 by jlucas9