nasa/nos3 #154: Sim Terminal Exception when Switching to UART Bus Type

msuder, 2023-03-23:
To reproduce:
Run:
nos3@nos3:~/Desktop/git/github.com/nasa/nos3-simterm-protocol-149/sims/build/bin$ ./nos3-single-simulator terminal

In SimTerminal:
SimTerm-default<sample-command>:(COMMAND)command:[IN=ASCII:OUT=ASCII] $ set simbustype uart
2023-03-23 12:27:28.194092 [ DEBUG ] - BaseConnection: deleting old base connection Handle
2023-03-23 12:27:28.194566 [ ERROR ] NosEngine.Common - receive operator failed to receive message, connection was shutdown in NosEngine::Transport::Connection::process_stop(std::unique_lock<std::mutex> &) at transport/src/Connection.cpp:377
2023-03-23 12:27:28.194661 [ ERROR ] NosEngine.Common - receive operator caught exception while receiving message, will stop receiving, message connection was shutdown, before async receive message started
2023-03-23 12:27:28.198200 [ ERROR ] NosEngine.Common - receive operator failed to receive message, connection was shutdown in NosEngine::Transport::Connection::process_stop(std::unique_lock<std::mutex> &) at transport/src/Connection.cpp:377
2023-03-23 12:27:28.198282 [ ERROR ] NosEngine.Common - receive operator caught exception while receiving message, will stop receiving, message connection was shutdown, before async receive message started
2023-03-23 12:27:28.221535 [ ERROR ] - SimTerminal::run:  Exception caught!
2023-03-23 12:27:28.222508 [ INFO  ] - main:  terminal simulator terminating

Unfortunately, no information on the exception caught is being reported :-(

jlucas9, 2025-03-26:
Backlog as switch to QEMU make make NOS Engine obsolete.

msuder, 2025-08-22:
This is still an issue and should remain open.