nasa/nos3 #158: Connection with FPrime & software simulation

ivanperez-keera, 2023-03-27:
Hello NOS3!

I posted a question in FPrime's discussions (https://github.com/nasa/fprime/discussions/1945) and someone indicated that:

> I did recently see a presentation about [NOS3](https://github.com/nasa/nos3) that indicated their project was intending to include F Prime as a FSW option. This would leverage Goddard's [42](https://github.com/ericstoneking/42) simulator as the frontend.

Has that been done? Would it be possible to access it?

jlucas9, 2023-03-27:
Hi @ivanperez-keera,
We do have it on the roadmap to assess and attempt to integrate as part of issue https://github.com/nasa/nos3/issues/84. We're supporting some teams at JPL who may be funding that effort to proceed in the more near term, but that hasn't begun just yet.

ivanperez-keera, 2023-03-27:
Thanks!

I'm then closing this as duplicate of #84.

closed 2023-03-27 by ivanperez-keera