nasa/nos3 #168: Mission Deployment

jlucas9, 2023-04-04:
Need to review and update the mission deployment suggestions and scripts provided for use.

zlynch2, 2025-04-02:
questions a new possible NOS3 user may ask? (where to find)
* What is NOS3? (read the docs)
* Why should NOS3 be used? (read the docs)
* When should NOS3 be used? Where does it fit in the traditional mission timeline? (read the docs)
* How to get get started with NOS3? Is there a demo or tutorial? (read the docs)
* How should I go from NOS3 as is to my specific mission?
* What is the development flow for creating a new component? (read the docs, with the closing of issue #548 )
* How to a move from software / simulator to hardware once it arrives?
* What can NOS3 give me during I&T? (read the docs)
* NOS3 uses during operation? (read the docs)

zlynch2, 2025-04-02:
wiki was updated to point to read the docs

zlynch2, 2025-04-07:
Will be updating markdown to include better component documentation. However, all the questions are answered in read the docs. 

Also, updated deployment repo dev branch to reflect readme on nos3 repo. Link below:
https://github.com/nasa-itc/deployment/tree/dev

Clsoing 168 as I verified all the information exists and wiki and readmes are correct.