nasa/nos3 #173: Mission Documentation

jlucas9, 2023-04-13:


kevincbruce, 2024-12-11:
Starting to look at different ways of generating documentation from in code. One I am familiar with is Doxygen ([Doxygen](https://www.doxygen.nl/index.html)), which generates HTML or similar documentation from comments (If I remember the comments need to be formatted in a particular way). They should link together based on variables and structs and the like, making it easy to dive through what stuff is. Also, it would encourage us to start putting descriptions in of the parameters of structs and headers for structs and methods to describe what they are. There may be issues with it diving all the way into cFS, as it would be quite labor intensive, so I'll also look at other possibilities, and post them here as I do, and then try to hold a tagup (hopefully before the holiday, or as a quick topic at the end of a weekly) to try and get some eyes on different possibilities.

jlucas9, 2024-12-19:
Was recommend we assess following:
* https://about.readthedocs.com/
* https://www.sphinx-doc.org/en/master/

jlucas9, 2025-03-20:
Break into smaller issues on a per document level.

jlucas9, 2025-04-10:
Merged into `dev` ^

PR #613 (2025-04-10): [nasa/nos3#554] STF ConOps
Rolled in information regarding passes and mission modes into a concept of operations document.
Also generally updated a few other things to better prioritize information in the READTHEDOCS.

Can view changes at https://nos3.readthedocs.io/en/nos3554-conops/ or directly in this PR.

Closes #173, #554, #556, and #559.
closed 2025-04-10 by jlucas9