nasa/nos3 #183: how to add generic_esp to the components

roypyp, 2023-05-22:
I want to simulate satellite health telemetry for my research. However, I need help understanding how to add generic_esp to the components of the nos3 simulator. I would like to know how to add that component and if there are other components to add to have more satellite health telemetry.

Thanks, Roy.

roypyp, 2023-05-25:
Thank you for the answer.

I would like to know if I understand it correctly; the components only work stand-alone, and if I add the generic_esp now, it will not connect to the other components.

So even if I add the generic_esp, it will not help to simulate the esp of the satellite because it is not connected to the other components?

jlucas9, 2023-05-25:
That's correct, the component applications only all talk to the simulated component directly and then publish that information on the cFS software bus. That data is essentially just sent through the generic radio to the ground software (COSMOS) and not used for any other internal processing on the spacecraft, but will be soon as the mission aspect comes together! 

Components can subscribe to other component messages and take action on them if desired on the spacecraft, but all data is also sent to the ground. It depends on where you want the research to apply (on the spacecraft or on the ground) as to how I'd recommend proceeding. Let me know in this thread and I'll see what I can suggest!

roypyp, 2023-05-31:
Thank you very much for the help.

I want to make sure I understand how this version work now. The version has components like the electrical power system (EPS), but they still work stand-alone from the other components?

Another question is when I try to look at the telemetry of the current vault and temperature. They stay the same no matter what kind of commend I send to the EPS component. Can I want to simulate a real-life satellite usage that changes the current vault or the temperature of the item of the ESP component and then try to create an anomaly. I need help understanding how to do it.

Is there a guide on how to use the SIM Terminal or UDP Terminal?

Thank you again for all the help!

jlucas9, 2023-06-01:
Hi @roypyp,
Let me break this down to try to answer your questions:
* Do the components like the EPS still work stand-alone from other components?
  - The EPS is a special case in that is out first attempt at simulator to simulator communications, on startup the EPS sends the SIM terminal commands to enable/disable depending on the configuration. When a switch is activated these same terminal commands get sent again. This was an attempt to get closer to reality in that if a component doesn't have power, it won't be able to communicate with the application.
  - Each generic "component" as we have them defined is a physical device and comes with a flight software application, simulation, and a ground database. These generic components don't really talk to one another, exception mentioned in above bullet. Following the data flow the cFS scheduler sends a housekeeping request at a set cadence to the application who communicates via the HWLIB to the simulation. The simulation interprets the request, pulls data from external providers like the 42 dynamics engine as needed, then generates a reply to the application. The application interprets the data, formats it into a packet to be published on the software bus. Anyone subscribed to this packet will be able to receive that data. An example of the next level of data processing and fusion can be followed in the Generic ADCS series of issues.
* EPS telemetries appear to remain unchanged
  - The EPS should change to show voltage and current based on it's configuration file depending on which switch is enabled. For example if you turn on [switch 1 the simulator configuration](https://github.com/nasa/nos3/blob/main/sims/cfg/nos3-simulator.xml#L218) shows it should read 1.23 volts and 4.56 amps. Note that these changes would need to be commanded currently as NOS3 doesn't have any limit checking or responses built in just yet.
  - Additional fidelity can certainly be added to the EPS, in an effort to generate a more complete mission quickly the generic components have been published in a lower fidelity state
* Do guides exist for SIM or UDP Terminal?
  - Nothing formal yet, the SIM / STDIO terminal should respond to `help` and print to the screen the commands available. These are not what is available within each simulator, but how to communicate with and switch between them. The generic simulators have help, enable, and disable commands supported minimally. By default the SIM terminal is setup to communicate with the SAMPLE simulator. You can do this by using the following command: `transact 120 help`. If you have the SAMPLE sim up at the same time you can see in the log it receives the command and sends the response back to the SIM terminal. 
  - The UDP terminal is the same thing as the SIM terminal, just accepting commands via a UDP port instead of the terminal directly
* What is the difference between generic_eps and generic_eps_radio?
  - A generic radio component has been added to the latest NOS3 release. The _radio to commands indicates that the command or telemetry flows through the radio component while the other uses the debug interface that wouldn't be available while in orbit (essentially plugged into the spacecraft).

Let me know if you have any other questions and have a great day!

jlucas9, 2023-06-13:
Closing for now due to age, if you have any other questions please feel free to follow up with a new issue!

closed 2023-06-13 by jlucas9