nasa/nos3 #195: Looking for COSMOS File Creator (CFC)

astrxlegaulois, 2023-06-19:
Hello,

I am looking for the tool COSMOS File Creator (CFC) that according to some NOS3 documentation used to be available under nos3/support/cfc.
http://stf1.com/NOS3Website/docs/NOS3_CombinedDocumentation.pdf
http://www.stf1.com/NOS3Website/docs/NOS3_Users_Guide_v1.04.pdf
I was unable to get the tool from any of the previous releases nor anywhere in the git tree.
I am inquiring because I suspect that this tool would be able to parse my cfe folders to generate a COSMOS configuration.
Are you able to push the CFC tool? Do you have any alternate to recommend?

I thank you in advance.

jlucas9, 2023-06-26:
Hi @astrxlegaulois,
After following up with some team mates it seems we never fully released CFC. This was due to CFC requiring special headers to be included with telemetry structures which was just as much work as writing the database by hand. We did experiment with and have success with CCDD - however it appears with the planned cFS upgrade currently underway XTCE will be the way to go. I found a reference here in the cFS discussions board - https://github.com/nasa/cFS/discussions/326

Sorry I don't have better news, but currently in NOS3 everything for the COSMOS 4.X databases is handwritten. Let me know if you have any other questions and have a great day!

jlucas9, 2024-03-19:
Closing for now, please re-open or create a new issue if you have any other questions!

closed 2024-03-19 by jlucas9