nasa/nos3 #196: Multiple Spacecraft Docker

jlucas9, 2023-06-26:
Continuation of https://github.com/nasa/nos3/issues/163 to get multiple spacecraft running together. 

Related issues:
- [ ] https://github.com/nasa/nos3/issues/139
- [ ] https://github.com/nasa/nos3/issues/140

jlucas9, 2023-08-25:
Looks to leverage docker compose for this instead of the previously used bash scripts

jlucas9, 2024-03-06:
## Initial drawing
[image]

Received the above from BJ in trying to better capture this effort. While previously we have the launcher script working to deploy multiple spacecraft (with minor manual edits needed to the simulator configuration file), however recently I've broken things with addition of the configuration (./cfg) folder and XML configuration files.

## End Goal
The mission configuration file and associated `./script/configure.py` that is run as part of  the `make config` step would prepare all the files necessary to run with a single or multiple spacecraft.

## Stepping stones
* Assess the existing `./scripts/docker_launch.sh` and how it was done previously
* Assess other options such as Compose / Swarm and Kubernetes to handle this
* Select deployment mechanism and get working
* Update `make config` to allow easy reconfiguration

While this effort is a high priority for the NOS3 development team I've added @Donnie-Ice  and @kevincbruce to this issue to start pushing it forward on our side as well!

bjkowalski, 2024-03-06:
Perhaps it is obvious to most, but to clarify the drawing, each "cloud" is meant to represent a Docker "bridge" network and one or more Docker containers directly attached to the "bridge".  Given my naive viewpoint on the internals of NOS3 and other components I created this to represent what I felt might be the desired endgame.

Below is a diagram of the current Docker environment as deployed by the scripts in the "dev" branch.
[image]

In the OpenC3 COSMOS cloud, the container "cosmos_openc3_cosmos-init_1" is shown as a dotted element as it is transient and exists only for a short period as the OpenC3 COSMOS instance is being created.

As currently deployed, 42 is a container within the Spacecraft.  Since my understanding of 42 is that it was intended to be the source of truth for all satelites in a constellation I assumed it was a single instance which is why it was shown outside the spacecraft in my original drawing.  Perhaps that is an incorrect assumption.

The current deployment has the NOS Time Driver as an external source.  Is it correct to assume that there would be only one for multiple SC?

In the Satellite cloud, I purposely illustrated some containers with solid lines and some with dotted lines.  My intent was to show those that are "optional" and those that would be required in all deployments.

The currently deployment wires the "cosmos_openc3-operator_1" container to the "nos3_sc_1" spacecraft network.  I would suggest that is a bad approach as it requires this operation be repeated for each spacecraft in the constellation.  I would suggest that a new "NOS3 Network/Radio" cloud be created which simulates the communications between the GSW and FSW components.

Isgaroth-the-Green, 2024-03-18:
Thank you for the diagram; it looks good. You mentioned a few points you were unsure about; at present, the following is true:

- Right now, we have an instance of 42 for each spacecraft. Whether that is likely to change will probably depend on how many spacecraft we have to simulate and how effective 42 is at simulating truth for several spacecraft; in the near future, though, I expect it to be one instance per spacecraft.
- You are correct that there will be only one NOS Time Driver regardless of the number of spacecraft. At present that is one of the obstacles to easy scalability of the system, as there are a few places where the spacecraft names/networks have to be hardcoded so that the NOS Time Driver can talk to all of them.

Regarding the way in which COSMOS is connected to the spacecraft, we used the connection between the "cosmos_openc3-operator_1" container and the "nos3_sc_1" network as the quickest and simplest way to connect the two. Also, we were working with a model in which COSMOS would only communicate to a single spacecraft, which would then talk to the next one down the line, and so on. How to change it in the future will depend somewhat on how the networking ends up; if we want the ground station to be able to communicate with each and every spacecraft, it might be wise to create another network for COSMOS to talk to the spacecraft. If we stick to each satellite talking only to the ones adjacent to it, though, another network might be superfluous.

How familiar are you with Docker Compose, and with container orchestration tools (like Kubernetes)? We are looking into using some kind of container orchestration tools to allow for much greater scalability than is possible on a single virtual machine, but none of us are particularly experienced in its use. Also, if you have knowledge on Docker networking more generally (and how it might be leveraged for this task) we are certainly interested in hearing it.

@kevincbruce and @Donnie-Ice are working on this issue; any of the three of us would be happy to answer questions to the best of our ability.

bjkowalski, 2024-03-19:
I have been developing and deploying Docker Containers and Docker Compose enabled solutions for a number of years.  Out of necessity I have had to dive fairly deep into Docker networking as well.   While I have dabbled with Kubernetes I haven't deployed it in a production environment.

My technical background is many decades of developing high performance computing, storage, and network solutions in both embedded and mainframe (a.k.a. supercomputing) environments.

I do have some thoughts on how the team might consider leveraging the capabilities that Docker networking and Compose provides which I will try to capture in a document and add to this thread later this week.

kevincbruce, 2024-03-19:
Thank you for the inputs, we look forward to reading the document and hearing some ideas on implementing a more efficiently scalable architecture!

bjkowalski, 2024-03-20:
Could someone elaborate on the scalability goals of the project?  I view the use of Docker Compose as a poor man's container orchestration capability.  With a single compose file one can easily stop/start a multi-container configuration, including networks, via a single command line.  While Kubernetes (K8s) and it's lighter peer K3s offer container orchestration across cloud and hybrid environments, I am not sure the added complexity warrants the investment.  I would have thought this project needs/wants the flexibility to have multiple mission software suites (gsw) and flight software suites (fsw) that may or may not be containerized deployed across a mixture of hardware configurations ranging from single server, to mutliple servers and perhaps even a hybred hardware environment.  Pursuit of a K8s (or similar) ecosystem would seem to limit the flexibility that is currently accessible.

Isgaroth-the-Green, 2024-03-20:
We are not entirely sure what the scalability goals will end up being; we know there is a desire, however, to have at least eight spacecraft running at the same time.

That makes sense about Docker Compose; my understanding is similar. My main reason for thinking some sort of container orchestration tool might be useful is twofold - firstly, I know Kubernetes and similar software allows for distributing containers across an arbitrary number of machines, which seems exceedingly difficult to do any other way. Secondly, if we were to deploy to the cloud, a Kubernetes architecture would seem both more convenient to deploy and also possibly cheaper to run (although I have not looked much at the cost of AKS vs. EC2 recently).

Regarding the first point, how would you suggest going about networking on multiple machines? Is that easier to do in Docker than I have previously imagined? You are right that this project should ideally have the flexibility to deploy across different hardware configurations, and it would be good to avoid the added complexity of Kubernetes, if possible.

bjkowalski, 2024-03-20:
If Cloud is in the future, then the investment in something like K8s is justifiable.  I would like to suggest that the team consider rather than attempting to have two significant architectural changes (adopting Kubenetes and adopting containers) perhaps there is an intermediate step.

The solution I was leaning towards may be unacceptable if you are adverse to using a "deprecated" capability that is still in production in many commercial organizations.  I was leaning toward an intermediate step where a Docker overlay network is used to create a multi-node "SpaceNet" (or name of your choice) which all Spacecraft (FSWs container instances) and Mission Control (GSWs) entities attach to rather than each other.  The overlay network leverages a Docker swarm capability which has been announced as deprecated but is still distributed within Docker, just not officially supported.  There are known performance issues with overlay networks but I am not sure if they will be detrimental to NOS3.  At a minimum this approach may provide insight into multi-node deployment that will inform the K8s approach decisions.  As Spacecraft are "launched", their Docker compose configuration attaches them to the overlay network via the Radio Sim.  Similarly the GSWs attach to the overlay network rather than directly to a Spacecraft.  This decoupling adds complexity for the simple case (single spacecraft <=> single GSW) but makes the architecture scale.  This approach also assumes that "time" is sourced from the "SpaceNet".

There is one HUGE assumption in all this, which exposes my ignorance of the existing implementation and Ground-Space communications in general.  I assume that GSW to FSW communications happens using CCSDS defined payloads transmitted via TCP/IP and that the "SpaceNet" becomes a CCSDS SLE like capability.  Another huge assumption is that there is some addressing implied in the CCSDS payloads.  I've included an updated initial diagram.  I haven't had time to dig into the exchanges between COSMOS and a Spacecraft to understand how that interface works with the existing software so my "assumption" could be totally off-base and my proposal was founded in vaporware.  If my assumptions are close to correct, the Overlay Network Docker cloud, shown in the middle, is the only component that becomes "node" aware - the only Docker setup that would need to know about multiple virtual or physical nodes and how to communicate between them.  All other elements would only need to understand that they need to talk to the overlay network and the overlay network takes care of delivering packets to the proper egress points.

Donnie-Ice, 2024-04-01:
Which feature specifically are you referring to as deprecated? I would like to do some research before we decide whether it is worth attempting to implement. Thanks in advance!

bjkowalski, 2024-04-02:
In my comments above I implied that Docker was dropping support for Overlay networks.  That was the context for my comment about using a deprecated feature.  In going back over my notes I found that what Docker has announced is dropping support for external key/value stores for overlay networks.  See the note on "Classic Swarm and overlay networks using cluster store" under URL https://docs.docker.com/engine/deprecated/#classic-swarm-and-overlay-networks-using-cluster-store. 

Since my oriinal comment above I have also become aware that overlay network encryption is not functional if Windows containers are part of the mix (URL = https://docs.docker.com/network/drivers/overlay/#encrypt-traffic-on-an-overlay-network).

kevincbruce, 2024-04-25:
We performed an investigation into different techniques, and came to the conclusion that, at least for now while we are first working with containerization. it is best to keep our current approach while we iron out bugs with performance and containers, then then upgrade to an orchestrated solution later when cost, performance, and scalability demand it, and when we are more confident working with such solutions.

With Docker Compose, we made attempts to upgrade the scripts to utilize Compose, however we found there would be limited benefit. First, a set of scripts or compose files spawn the containers for COSMOS (4 or 5) separately from this. Then, once each spacecraft network is created, it must be linked to the COSMOS containers before anything else can be done. This necessitates creating the spacecraft networks outside of Compose, as you cannot connect an external container to an internal network within compose. Then, beyond this issue, we found that we would need to spawn and connect things in a different order in Compose when compared with the current, which could make things make less sense and could lead to issues and glitches. Finally, we want all container windows to show up in the terminal so we can monitor and control things. This would require us to run 'docker exec' commands for each container after the Compose anyways. All of these factors combined defeat the purpose of utilizing Compose - decreasing the number of commands in our scripts while maintaining similar operations - entirely, unless we move to full Orchestration.

Looking into Docker Swarm, it seemed that it requires all containers to work as part of one big cluster. This does not work well for our purposes, as each separate app, sim, or component for each spacecraft is a separate container. Each of these must work independently from each other, and we will likely need separate spacecraft working as separate entities in some applications, thus Swarm is likely too restrictive on individual components for our purposes.

Thus, we also looked into Kubernetes, as it seemed to be the better orchestration solution for our purposes, as it is more configurable in terms of how independent each node acts and how nodes are grouped. While Kubernetes, in theory, is an attractive solution to our problems, and may well be implemented as cloud solutions get further along, we've found that as of now, due to our lack of experience with orchestration and the complexity of the solution, it is likely more effort than it is worth at this stage, while we are still working on getting the current level of containerization working and squashing performance and time driver bugs that are exacerbated by having multiple spacecraft (noted in the issue nos3#285, if you'd like to help investigate).

Due to these factors in our investigation, along with the fact that with the 42 GUI disabled and using COSMOS 4 each satellite seems fairly lightweight, we have decided it is prudent to continue perfecting containerized performance with our current script setup. We will seek to fix performance issues, better understand and optimize containerization in Docker, and figure out how best to control the multiple spacecraft (one instance of COSMOS 4 with targets for each spacecraft's debug and radio channels and targets for each spacecrafts' instances of each app, one instance of COSMOS 4 for each spacecraft, or utilizing COSMOS 5 if it offers better multiple spacecraft support to the point where it justifies the extreme performance losses). Then, later on, once we are more confident with containerization and have our current solution optimized, we will look into orchestration again (likely through Kubernetes) due to its benefits in terms of cost and scalability on the cloud.

Thank you for your assistance and suggestions in the matter! It aided our investigation and improvements throughout the process, and we hope to see more suggestions in the future as we perfect our current solution, and possibly move to orchestration in the future.

closed 2024-05-16 by jlucas9