nasa/nos3 #210: NOS3 OSAL - OS Task Delay

jlucas9, 2023-08-03:
Looking at HTOP it appears some applications are using way more processing time than others:
[image]

These three applications (NAV_CHILD_TASK, GENERIC_RADIO, and TO_LAB_APP) all use OS_TaskDelay to drive the application loop functionality. Tracking this down I think the `fsw/osal/src/os/nos/src/NOS-Time.c > NOS_clock_nanosleep` function could be the culprit.

jlucas9, 2023-08-03:
I can confirm this appears to the be the right location, if I change `OS_TaskDelay_Impl` form NOS3 to what is in POSIX there are no problems.

jlucas9, 2023-10-31:
# Debugging OS Task Delay

Commit [c36ee4b], Branch `nos3#215-Local-NOS3`
* Baseline
  * 170-180% utilization of core-cpu1
* Posix fix wholesale 
  * 2.0-2.7%$ utilization of core-cpu1
* Only change `clock_nanosleep`
  * 170-180%
* Posix, but divite sleep by 64
  * 4.0-6.0%
* Posix, but divide sleep by 2048
  * 20-25%
* Posix, but divide sleep by 16x
  * Run at 16x time
  * 42 now over 100% (no graphics)
  * Re-test
    * 1x, good
    * 2x, good
    * 4x, good
    * 8x, Msg Limit Err occured in FSW terminal, 42 over 100% CPU utilization

jlucas9, 2023-11-07:
What in NOS3 uses `OS_TaskDelay`?
* Apps
  * CF, semaphore task delay
  * FM, CopyInProgress / GettingCRC / getSizeTimeMode loop delay
  * TO_Lab, main run loop
* cFS
  * cFE, various tests
  * ES, CFE_ES_PANIC_DELAY and MainTaskSyncDelay
  * EVS, CFE_EVS_PANIC_DELAY
  * OSAL, various tests
  * PSP, exit and restart delays
* Components
  * Arducam, child task loop
  * GPS, child task loop
  * HWLIB, uart_read_port delay between character reads
  * Radio, child task loop
  * Sample, uart bytes available timeout loop

How does SCH sync time?
* TimeSemaphore

msuder, 2023-11-09:
Experimentation seems to point to division being the cause of slowness in NOS_clock_gettime.  It is probably the cause of slowness in NOS_clock_nanosleep also.

dccutrig, 2023-11-11:
I'm not 100% convinced I've replicated this on my end, but some useful testing may have still come out of it:

Within OS_TaskDelay_Impl:
- NOS_clock_gettime and NOS_clock_nanosleep: ~80%
- NOS_clock_nanosleep and  clock_gettime: core_cpu1 cpu utilization:  ~75%
- clock_nanosleep  and NOS_clock_gettime: core_cpu1 cpu utilization: ~84%
- clock_nanosleep and clock_gettime: ~2% with occasional spikes

I tried running valgrind drd and callgrind, but not being 100% I recreated the issue those won't be as useful.

This might be useful to profile with valgrind for one of you guys:

_valgrind --tool=callgrind -v --dump-every-bb=10000000 ./core-cpu1 --reset PO_

Callgraphs can statistics can be viewed if you install kcachegrind

Isgaroth-the-Green, 2023-11-30:
Why is the NOS_clock_nanosleep code set up to call clock_nanosleep so many times? I assume there is a reason, but I am not sure at present what it is. I tried changing it to only call clock_nanosleep once per loop (with its duration being the whole quantity of time) and it seems to be somewhat less memory-intensive, though I have not tested that quantitatively.

jlucas9, 2023-12-05:
I think it was done to accommodate possible drift in sim time delays so that we don't over sleep - it also could help if the time tic is speed up during the sleep.

dgeletko, 2024-01-16:
I just want to post a comment describing what we discovered today. If using any delay/sleep call, you have to use relative time with NOS time. If using wall clock time, it doesn't matter, however, you probably shouldn't be using wall time. The issue is that if you use absolute time, but it is an absolute time from the NOS time (i.e.; from `NOS_clock_gettime` ), it most likely will never result in an actual sleep but will return immediately, effectively 'pegging' the CPU. This is simply because the 'epoch' or 'base' values of wall vs NOS time are completely different. The wall time is most likely a much larger value.

jlucas9, 2024-01-16:
https://github.com/nasa-itc/osal/pull/3

jlucas9, 2024-01-18:
Merged to `dev`

closed 2024-01-18 by jlucas9