nasa/nos3 #216: Component CI Pipeline

jlucas9, 2023-09-25:
Can leverage examples from https://github.com/nasa/sample_app

Will need to update our component sample app.

Isgaroth-the-Green, 2023-12-14:
I believe the next step is to check that the sample_app works within cFS and then compare it to the sample in NOS3 (at which point the CI tests in sample_app can be leveraged); I have not been able to get cFS compiling properly, though. It always gives an error regarding stubs, approximately
`/usr/bin/ld: cannot find -lut_core_api_stubs: No such file or directory`
and 
`/usr/bin/ld: cannot find -lcoverage-sample_lib-stubs: No such file or directory`
The command (I think) which gives this error is
`/usr/bin/cc -g CMakeFiles/coverage-sample_app-sample_app_uitls-testrunner.dir/coveragetest/coveragetest_sample_app_utils.c.o CMakeFiles/coverage-sample_app-sample_app_utils-object.dir/home/jstar/Documents/code/cFS/apps/sample_app/fsw/src/sample_app_utils.c.o -o coverage-sample_app-sample_app_utils-testrunner -pg --coverage -lut_core_api_stubs ../../../osal/ut_assert/libut_assert.a -lcoverage-sample_lib-stubs libcoverage-sample_app-stubs.a libcoverage-sample_app-stubs.a libsample_app_ut_common.a ../../../osal/ut_assert/libut_assert.a ../../../osal/libosal_bsp.a -lpthread -ldl -lrt`

jlucas9, 2024-06-13:
Opted to just do NOS3 level CI and not with each component individually. So when you commit to the top level of NOS3 with the new component submodule it'll kick off, but not before then.

jlucas9, 2024-06-14:
Merged and squashed to `dev`

closed 2024-06-14 by jlucas9