nasa/nos3 #229: WSL Deployment

jlucas9, 2023-11-30:
Document deployment to the Windows Subsystem for Linux (WSL).

sdunlap-afit, 2024-06-11:
I'm new to NOS3, but I've been playing with it on Windows with WSL and Docker Desktop. Everything seems to work, except for one issue. Due to a known issue with Docker on Windows (and Mac?), `--ulimit rtprio=99 --cap-add=sys_nice` does not work in Docker run commands. This causes the flight software to abort. It may be causing other issues as well, but I'm not familiar enough to spot any.

https://github.com/moby/moby/issues/38631

The only workaround I've found is to set `OSAL_CONFIG_DEBUG_PERMISSIVE_MODE` true in `nos3/fsw/osal/default_config.cmake`. This allows the flight software to continue despite failing to set thread priority.

I just wanted to pass this on in case it helps someone in the future.

FSW Output:

```bash
CFE_PSP: Reset Type: PO
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 42
CFE_PSP: Default CPU Name: cpu1
OS_Posix_GetSchedulerParams():191:Policy 1: available, min-max: 1-99
OS_Posix_GetSchedulerParams():191:Policy 2: available, min-max: 1-99
OS_Posix_TaskAPI_Impl_Init():377:Selected policy 2 for RT tasks, root task = 99
OS_Posix_TaskAPI_Impl_Init():394:Could not setschedparam in main thread: Operation not permitted (1)
OS_API_Init():146:OS_API_Impl_Init(0x1) failed to initialize: -1
OS_API_Init():226:Warning: Microsecs per sec value of 0 does not equal 1000000 (MicroSecPerTick: 0   TicksPerSecond: 0)
CFE_PSP: OS_API_Init() failure
OS_printf():290:BUG: OS_printf() called when OSAL not initialized: CFE_PSP_Panic Called with error code = 0x%08X. Exiting.
OS_printf():290:BUG: OS_printf() called when OSAL not initialized: The cFE could not start.
./scripts/fsw_respawn.sh: line 26: 10947 Aborted                 $FSW_DIR/core-cpu1 -R PO
```

jlucas9, 2025-03-17:
It doesn't seem that https://github.com/nasa/nos3/blob/main/cfg/nos3_defs/native_osconfig.cmake#L39 is used. Need to set permissive mode in https://github.com/nasa/nos3/blob/main/cfg/nos3_defs/default_osconfig.cmake

jlucas9, 2025-03-31:
Merged into `dev` ^

closed 2025-03-31 by jlucas9